txoof/epd_display #2: AttributeError: 'EPD' object has no attribute 'display'

ZubaZ21, 2021-04-09:
Raspberry Pi 3 Model B Rev 1.2
Buster
Waveshare 3in7

I am able to run the waveshare demo

I have disabled all plugins so the only thing changed is the display_type

[main]
# waveshare display type
display_type = epd3in7

And yet, when I run the executable . . . failure:

$ ./paperpi
Traceback (most recent call last):
  File "paperpi/paperpi.py", line 584, in <module>
  File "paperpi/paperpi.py", line 546, in main
  File "paperpi/paperpi.py", line 293, in setup_display
  File "epdlib/Screen.py", line 251, in epd
AttributeError: 'EPD' object has no attribute 'display'
[2366] Failed to execute script paperpi

[It's probably me.  Sorry.]

txoof, 2021-04-10:
@ZubaZ21 

It looks like the 3in7 [display does not conform to the same standards](https://github.com/waveshare/e-Paper/blob/af9694c602fb24d776cd0f0b3e073a477294e3d8/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd3in7.py#L388) as the other displays. I should have caught this when putting together the documentation. 

I'll see if there's a way to patch around this and fix this up without too much of a kludge. I have a few ideas. Would you be willing to run some tests for me once I have something that might work?

ZubaZ21, 2021-04-10:
Of course!

txoof, 2021-04-11:
@ZubaZ21 
I think I have a solution; I've done a bit of a kludge on the base library that runs this to try and fix the issue.

The [tgz linked here](https://github.com/txoof/epdlib/raw/grayscale/Screen.tgz) can be executed with `./Screen` 

When prompted, enter your screen type to run a demo. If all goes well, you should four different versions of the same text displayed sequentially in different orientations. If it fails, share the debug output here and we can go from there.

ZubaZ21, 2021-04-11:
Output was displayed.
Of note (and it may be as intended but I don't know clearly what was being tested): The screen was not cleared before your test ran or between the changes or orientation.

Lastly, thanks for finding an issue and not making me feel like a damn fool.

[image]

txoof, 2021-04-11:
@ZubaZ21: Well, that's progress!

It should have displayed the text in four different orientations, clearing the screen between each redraw.

I suspect the 3in7 screen also has a weird clear function from the other screens. I suspect I can sort this out too. Can you share the debug output (all of it)? 

Thanks for testing this out. It's super helpful to get feedback. Sometimes folks leave trouble tickets and never follow up so the bug just remains open forever.

ZubaZ21, 2021-04-11:
Happy to help.

For what it's worth, the demo provided by waveshare clears as expected.

Ummm... is the debug write to a log? Or what's the best way to output? 
(Or I can just copy/pasta?)

txoof, 2021-04-11:
Yeah, you can just copy paste from the terminal output. There's probably several screens worth, but I'll take it all. You can make it easier to read by surrounding it with three back ticks (```) before and after. That will make it fixed width and easier for me to read.

```
It will look like this when you post the comment.
```

In other news, the reason my code doesn't work properly is because the waveshare libraries are **GARBAGE**. The clear method has two unneeded arguments in it that are causing all the problems for your screen.

I think I've found a quick fix that should improve things for all the screen types. 

I'll work on building another test bed around that and share it with you shortly. 

If you're on discord, we can continue here: 
https://discord.gg/mpACrDJY

txoof, 2021-04-11:
[Here's an updated version of the demo](https://github.com/txoof/epdlib/raw/grayscale/Screen.tgz)

Once we get that ironed out, I can build paper_pi against the library and push out a version that will hopefully work for you.

ZubaZ21, 2021-04-11:
Latest build.  No love (or rather, the same love); no clearing of things.

```
Enter screen type to run test/demo: epd3in7
setup for rotation: 0
DEBUG:root:Horizontal
INFO:root:using alternative EPD display method (display_1Gray) for this screen
DEBUG:root:Screen rotation set to: 0
DEBUG:root:creating layout
DEBUG:root:calculating layout for resolution [480, 280]
DEBUG:root:layout id(1972730448)
DEBUG:root:*****title*****
DEBUG:root:checking layout keys
DEBUG:root:adding key: padding: 0
DEBUG:root:adding key: rand: False
DEBUG:root:adding key: inverse: False
DEBUG:root:adding key: maxchar: None
DEBUG:root:adding key: dimensions: None
DEBUG:root:adding key: scale_x: None
DEBUG:root:adding key: scale_y: None
DEBUG:root:dimensions: (480, 160)
DEBUG:root:section has absolute coordinates
DEBUG:root:coordinates: (0, 0)
DEBUG:root:calculating maximum font size for area: (480, 160)
DEBUG:root:using font: /tmp/_MEIob4FsK/fonts/Font.ttc
DEBUG:root:target X font dimension 564.7058823529412
DEBUG:root:target Y font dimension 40.0
DEBUG:root:Y target reached
DEBUG:root:test string: W W W ; pixel dimensions for fontsize 42: (147, 41)
DEBUG:root:*****artist*****
DEBUG:root:checking layout keys
DEBUG:root:adding key: padding: 0
DEBUG:root:adding key: rand: False
DEBUG:root:adding key: inverse: False
DEBUG:root:adding key: font_size: None
DEBUG:root:adding key: maxchar: None
DEBUG:root:adding key: dimensions: None
DEBUG:root:adding key: scale_x: None
DEBUG:root:adding key: scale_y: None
DEBUG:root:dimensions: (480, 120)
DEBUG:root:section has calculated position
DEBUG:root:coordinates: (0, 0)
DEBUG:root:calculating maximum font size for area: (480, 120)
DEBUG:root:using font: /tmp/_MEIob4FsK/fonts/Font.ttc
DEBUG:root:target X font dimension 564.7058823529412
DEBUG:root:target Y font dimension 45.0
DEBUG:root:Y target reached
DEBUG:root:test string: W W W ; pixel dimensions for fontsize 48: (168, 46)
DEBUG:root:***title***)
INFO:root:set text block: title
INFO:root:TextBlock created
DEBUG:root:no maxchar set
DEBUG:root:setting old_font = None
DEBUG:root:calculating maximum characters for font ('WenQuanYi Micro Hei', 'Regular')
DEBUG:root:maximum characters per line: 21
DEBUG:root:formatting string: .
DEBUG:root:formatted list:
 ['.']
DEBUG:root:creating blank image area: (480, 160) with inverse: False
DEBUG:root:line size: 11, 41
DEBUG:root:max x dim so far: 11
DEBUG:root:dimensions of text portion of image: (11, 41)
DEBUG:root:hcenter line: .
DEBUG:root:drawing text at 0, 0
DEBUG:root:with dimensions: 11, 41
DEBUG:root:pasting text portion at coordinates: 234, 60
DEBUG:root:***artist***)
INFO:root:set text block: artist
INFO:root:TextBlock created
DEBUG:root:no maxchar set
[... 441 lines cut ...]
DEBUG:root:concating blocks into single image
DEBUG:root:pasitng **title** image at: (0, 0)
DEBUG:root:pasitng **artist** image at: (0, 160)
DEBUG:root:preparing to write to EPD
DEBUG:root:Horizontal
DEBUG:root:init and write to EPD
DEBUG:root:init screen -- grayscale: True
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x758e2d50> intialized
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
refresh screen -- screen should flash and be refreshed
DEBUG:root:init screen -- grayscale: True
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x758e2d50> intialized
DEBUG:root:There is no such mode
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
```

ZubaZ21, 2021-04-11:
v0.3.0
```
version: 0.3.0
My Path is: /tmp/_MEILlEnlB
Demo/Test of epdlib

Available Screen Types in waveshare_epd library:
epd1in02
epd1in54
epd1in54_V2
epd1in54b
epd1in54b_V2
epd1in54c
epd2in13
epd2in13_V2
epd2in13b_V3
epd2in13bc
epd2in13d
epd2in66
epd2in66b
epd2in7
epd2in7b
epd2in7b_V2
epd2in9
epd2in9_V2
epd2in9b_V3
epd2in9bc
epd2in9d
epd3in7
epd4in01f
epd4in2
epd4in2b_V2
epd4in2bc
epd5in65f
epd5in83
epd5in83_V2
epd5in83b_V2
epd5in83bc
epd7in5
epd7in5_HD
epd7in5_V2
epd7in5b_HD
epd7in5b_V2
epd7in5bc
epdconfig
Enter screen type to run test/demo: epd3in7
setup for rotation: 0
DEBUG:root:Horizontal
INFO:root:using alternative EPD display method (display_1Gray) for this screen
DEBUG:root:Screen rotation set to: 0
DEBUG:root:creating layout
DEBUG:root:calculating layout for resolution [480, 280]
DEBUG:root:layout id(1971919344)
DEBUG:root:*****title*****
DEBUG:root:checking layout keys
DEBUG:root:adding key: padding: 0
DEBUG:root:adding key: rand: False
DEBUG:root:adding key: inverse: False
DEBUG:root:adding key: maxchar: None
DEBUG:root:adding key: dimensions: None
DEBUG:root:adding key: scale_x: None
DEBUG:root:adding key: scale_y: None
DEBUG:root:dimensions: (480, 160)
DEBUG:root:section has absolute coordinates
DEBUG:root:coordinates: (0, 0)
DEBUG:root:calculating maximum font size for area: (480, 160)
DEBUG:root:using font: /tmp/_MEILlEnlB/fonts/Font.ttc
DEBUG:root:target X font dimension 564.7058823529412
DEBUG:root:target Y font dimension 40.0
DEBUG:root:Y target reached
[... 519 lines cut ...]
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x7581cd90> intialized
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
refresh screen -- screen should flash and be refreshed
DEBUG:root:init screen -- grayscale: True
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x7581cd90> intialized
DEBUG:root:There is no such mode
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:init screen -- grayscale: True
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x7581cd90> intialized
DEBUG:root:There is no such mode
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
```

ZubaZ21, 2021-04-11:
```

version: 0.3.1
My Path is: /tmp/_MEIiCTZve
Demo/Test of epdlib

Available Screen Types in waveshare_epd library:
epd1in02
epd1in54
epd1in54_V2
epd1in54b
epd1in54b_V2
epd1in54c
epd2in13
epd2in13_V2
epd2in13b_V3
epd2in13bc
epd2in13d
epd2in66
epd2in66b
epd2in7
epd2in7b
epd2in7b_V2
epd2in9
epd2in9_V2
epd2in9b_V3
epd2in9bc
epd2in9d
epd3in7
epd4in01f
epd4in2
epd4in2b_V2
epd4in2bc
epd5in65f
epd5in83
epd5in83_V2
epd5in83b_V2
epd5in83bc
epd7in5
epd7in5_HD
epd7in5_V2
epd7in5b_HD
epd7in5b_V2
epd7in5bc
epdconfig
Enter screen type to run test/demo: epd3in7
setup for rotation: 0
DEBUG:root:Horizontal
INFO:root:using alternative EPD display method (display_1Gray) for this screen
DEBUG:root:Screen rotation set to: 0
DEBUG:root:creating layout
DEBUG:root:calculating layout for resolution [480, 280]
DEBUG:root:layout id(1972709968)
DEBUG:root:*****title*****
DEBUG:root:checking layout keys
DEBUG:root:adding key: padding: 0
DEBUG:root:adding key: rand: False
DEBUG:root:adding key: inverse: False
DEBUG:root:adding key: maxchar: None
DEBUG:root:adding key: dimensions: None
DEBUG:root:adding key: scale_x: None
DEBUG:root:adding key: scale_y: None
DEBUG:root:dimensions: (480, 160)
DEBUG:root:section has absolute coordinates
DEBUG:root:coordinates: (0, 0)
DEBUG:root:calculating maximum font size for area: (480, 160)
DEBUG:root:using font: /tmp/_MEIiCTZve/fonts/Font.ttc
DEBUG:root:target X font dimension 564.7058823529412
DEBUG:root:target Y font dimension 40.0
DEBUG:root:Y target reached
[... 514 lines cut ...]
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x758dddb0> intialized
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
refresh screen -- screen should flash and be refreshed
DEBUG:root:init screen -- grayscale: True
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x758dddb0> intialized
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:init screen -- grayscale: True
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x758dddb0> intialized
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
```

txoof, 2021-04-12:
@ZubaZ21 
Making some progress. New version available in the grayscale channel -- https://github.com/txoof/epdlib/raw/grayscale/Screen.tgz

ZubaZ21, 2021-04-12:
I didn't expect to hear from you so soon.

Report:

- No initial clear
- Text is HUGE and doesn't fit the screen
- No clearing between rotations
- Text appears gray/grey

Video: https://photos.app.goo.gl/RgtHw8mEEe9Wuovv8

Debug:
```

version: 0.3.2_2020.04.12
My Path is: /tmp/_MEIoG418p
Demo/Test of epdlib

Available Screen Types in waveshare_epd library:
epd1in02
epd1in54
epd1in54_V2
epd1in54b
epd1in54b_V2
epd1in54c
epd2in13
epd2in13_V2
epd2in13b_V3
epd2in13bc
epd2in13d
epd2in66
epd2in66b
epd2in7
epd2in7b
epd2in7b_V2
epd2in9
epd2in9_V2
epd2in9b_V3
epd2in9bc
epd2in9d
epd3in7
epd4in01f
epd4in2
epd4in2b_V2
epd4in2bc
epd5in65f
epd5in83
epd5in83_V2
epd5in83b_V2
epd5in83bc
epd7in5
epd7in5_HD
epd7in5_V2
epd7in5b_HD
epd7in5b_V2
epd7in5bc
epdconfig
Enter screen type to run test/demo: epd3in7
setup for rotation: 0
INFO:root:reeassigining `display` method to `display_1Gray` for this screen
INFO:root:reassigning `getbuffer` method to `getbuffer_4Gray` for this screen
DEBUG:root:Horizontal
DEBUG:root:Screen rotation set to: 0
DEBUG:root:creating layout
DEBUG:root:calculating layout for resolution [480, 280]
DEBUG:root:layout id(1972623952)
DEBUG:root:*****title*****
DEBUG:root:checking layout keys
DEBUG:root:adding key: padding: 0
DEBUG:root:adding key: rand: False
[... 498 lines cut ...]
print some text on the display
DEBUG:root:concating blocks into single image
DEBUG:root:pasitng **title** image at: (0, 0)
DEBUG:root:pasitng **artist** image at: (0, 160)
DEBUG:root:preparing to write to EPD
DEBUG:root:Horizontal
DEBUG:root:init and write to EPD
DEBUG:root:init screen -- grayscale: True
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x758c9d50> intialized
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
refresh screen -- screen should flash and be refreshed
DEBUG:root:init screen -- grayscale: True
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x758c9d50> intialized
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
```

txoof, 2021-04-12:
I had a flash of insight while I was eating lunch: my insight was different wrong. Back to reverse engineering the craptacular driver file. 

I'm beginning to think it might just be easier to rewrite the entire driver stack and start from scratch. The waveshare libraries are a maze of unused arguments, poorly named functions and duplicate chunks of code.

This will be a fun project...

txoof, 2021-04-14:
@ZubaZ21 
I've started building my own set of drivers based on the spec sheets. This is helping me have a much better understanding of how the drivers work so I can hopefully trouble shoot this problem.

There's a **lot** of copy/pasta going on leading to redundant chunks of code in the drivers and stuff that's got fat-fingered once and has propagated through all the drivers. Or, the spec sheets are wrong. Equally possible that WaveShare maintainers don't read their own spec sheets, so that adds a whole fun level of 

 It could be a ***very*** long road before I manage to get to the 3in7 rewritten. I hope this doesn't dash your hopes completely.

ZubaZ21, 2021-04-14:
I. AM. FURIOUS.  ;)

Good luck.  Have fun.  Learn all the things.

Let me know how I can help.

txoof, 2021-04-18:
@ZubaZ21 
I've learned a lot, and maybe have an updated demo that *might* be ready to go.

Can you give it another try?

ZubaZ21, 2021-04-18:
Did not clear on run nor between rotations.  Did clear at end.

```

version: 0.3.3_2020.04.17
My Path is: /tmp/_MEIjxGn4K
Demo/Test of epdlib

Available Screen Types in waveshare_epd library:
epd1in02
epd1in54
epd1in54_V2
epd1in54b
epd1in54b_V2
epd1in54c
epd2in13
epd2in13_V2
epd2in13b_V3
epd2in13bc
epd2in13d
epd2in66
epd2in66b
epd2in7
epd2in7b
epd2in7b_V2
epd2in9
epd2in9_V2
epd2in9b_V3
epd2in9bc
epd2in9d
epd3in7
epd4in01f
epd4in2
epd4in2b_V2
epd4in2bc
epd5in65f
epd5in83
epd5in83_V2
epd5in83b_V2
epd5in83bc
epd7in5
epd7in5_HD
epd7in5_V2
epd7in5b_HD
epd7in5b_V2
epd7in5bc
epdconfig
Enter screen type to run test/demo: epd3in7
setup for rotation: 0
INFO:root:reeassigining `display` method to `display_1Gray` for this screen
DEBUG:root:Horizontal
DEBUG:root:Screen rotation set to: 0
DEBUG:root:creating layout
DEBUG:root:calculating layout for resolution [480, 280]
DEBUG:root:layout id(1971849856)
DEBUG:root:*****title*****
DEBUG:root:checking layout keys
DEBUG:root:adding key: padding: 0
DEBUG:root:adding key: rand: False
DEBUG:root:adding key: inverse: False
DEBUG:root:adding key: maxchar: None
DEBUG:root:adding key: dimensions: None
DEBUG:root:adding key: scale_x: None
DEBUG:root:adding key: scale_y: None
DEBUG:root:dimensions: (480, 160)
DEBUG:root:section has absolute coordinates
DEBUG:root:coordinates: (0, 0)
DEBUG:root:calculating maximum font size for area: (480, 160)
DEBUG:root:using font: /tmp/_MEIjxGn4K/fonts/Font.ttc
DEBUG:root:target X font dimension 564.7058823529412
[... 484 lines cut ...]
print some text on the display
DEBUG:root:concating blocks into single image
DEBUG:root:pasitng **title** image at: (0, 0)
DEBUG:root:pasitng **artist** image at: (0, 160)
DEBUG:root:preparing to write to EPD
DEBUG:root:Horizontal
DEBUG:root:init and write to EPD
DEBUG:root:init screen -- grayscale: True
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x7580bdb0> intialized
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
refresh screen -- screen should flash and be refreshed
DEBUG:root:init screen -- grayscale: True
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:<waveshare_epd.epd3in7.EPD object at 0x7580bdb0> intialized
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
```

txoof, 2021-04-18:
I'm gonna have to call uncle on this one. Without a 3in screen to test on, I'm just making guesses at this point. 

In a fit of frustration, I even looked around to buy one, but couldn't find one for sale! 

If you're interested in an upgrade, the 5in and 7in models are really nice and I've verified that they work with the EPD library that backend this project.

[image]

ZubaZ21, 2021-04-19:
Understood.

If you are ever interested, this is the one I picked up.

https://www.waveshare.com/product/displays/e-paper/3.7inch-e-paper-hat.htm

Maybe this is how I learn python and drivers. 😂

It was good to meet a fellow Aaron.

Aaron
www.todaysagoodday.com

txoof, 2021-04-19:
@ZubaZ21 
If you have some programming experience, python is ***dead easy*** to learn. If you're new to programming, python is a great language to start with too. 

If you want to give it a go and have a fiddle with the epdlib, this would be an interesting project.  There's nothing too complex going on in the epd library. I can coach you through the important parts.

The WaveShare demo code is documented reasonably well so you can get a feel for what's going on. It's a good place to start playing around too.

ZubaZ21, 2021-04-22:
See also: https://github.com/TomWhitwell/SlowMovie/issues/68 :)

txoof, 2021-04-24:
Now you've got an even better reason to buy a different model...

txoof, 2021-07-03:
@ZubaZ21 
I've been hacking away at this for a while and I've determined that the epd3in7 is an odd-ball display that doesn't play by the same rules as the rest of the waveshare line.

I think it would be pretty easy to re-write the epd module if you wanted to dig into the spec sheet and reverse engineer the other panel modules. I started to do this, but with no 3.7 to test against, it was pretty hopeless to make meaningful progress.

For now, I['ve tagged the 3in7 display as incompatible ](https://github.com/txoof/epd_display/tree/new_epdlib#supported-screens )and narrowed down the problem to the `Clear` and `display` functions.
```
* epd3in7 supported: False
 - unsupported EPD.Clear() function
- AttributeError: module does not support EPD.display()
```

Let me know if you want some support in hacking out your own module for this. It would be an interesting learning experience :)

closed 2021-07-03 by txoof