txoof/epd_display #19: screen does not clear properly when daemon is stopped

txoof, 2021-11-01:
@jhaake2308
Screen does not clear when daemon is stopped through systemctl. Systemd may be killing the process before it has a chance to complete cleanup.

txoof, 2021-11-15:
Anyone with Systemd experience that can look at this would be greatly appreciated.

txoof, 2021-11-15:
Try this https://stackoverflow.com/questions/18499497/how-to-process-sigterm-signal-gracefully

txoof, 2021-11-16:
@jhaake I think I've resolved this as well as issue #20 (ignoring max_refresh). Would you mind git pulling the pre_release branch and testing it out?

I've pushed a frozen executable [here](https://github.com/txoof/epd_display/raw/pre_release/paperpi_latest.tgz)

You can also clone the entire repo. and give it a go against your current plugin build.

jhaake2308, 2021-11-16:
cloning the pre_release right now, will give feedback asap

jhaake2308, 2021-11-16:
I was able to clone the repo and build paperpy with my plugin w/o errors. systemctl stop clears the screen right now, reproducable (doing it the wird time now ;) ). Also when shutting the pi down, it clears the screen. 

The variable from #20 seems also to be honored with this release. No more refreshs 
I am having no errors in syslog. 

Anything more you want me to test?

txoof, 2021-11-16:
That's great news!

I've got it going on a test setup as well. I'll let it run for a few days before I call it good and merge it into the main branch.

Thanks for testing.

jhaake2308, 2021-11-20:
seems to be running fine, no errors or crashes here

txoof, 2021-11-20:
Great. I've had a similar experience on my test setup.

I'll merge the pre_release branch into the main branch shortly.

closed 2021-11-20 by txoof