txoof/epd_display #32: xkcd plugin crashes when backend is not available

txoof, 2022-01-12:
looks like wrapping json decode in a try/except will resolve this
```
19:01:34 paperpi:update_plugins:454:INFO - __________Plugin: XKCD Comic Plugin__________
19:01:34 xkcd_comic:update_function:107:DEBUG - XKCD CONFIG: {'plugin': 'xkcd_comic', 'max_x': 800, 'max_y': 600, 'resize': 1, 'max_retries': 10}
19:01:41 xkcd_comic:get_comic_json:45:ERROR - failed to decode JSON result possibly due to previous errors: [Errno Expecting value]
<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN"
 "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
<html>
  <head>
    <title>503 Backend unavailable, connection timeout</title>
  </head>
  <body>
    <h1>Error 503 Backend unavailable, connection timeout</h1>
    <p>Backend unavailable, connection timeout</p>
    <h3>Guru Mediation:</h3>
    <p>Details: cache-ams12725-AMS 1642010501 2485529884</p>
    <hr>
    <p>Varnish cache server</p>
  </body>
</html>
: 1
Traceback (most recent call last):
  File "paperpi/paperpi.py", line 729, in <module>
  File "paperpi/paperpi.py", line 686, in main
  File "paperpi/paperpi.py", line 506, in update_loop
  File "paperpi/paperpi.py", line 459, in update_plugins
  File "/tmp/_MEIqImArp/library/Plugin.py", line 254, in update
    is_updated, data, priority = self.update_function(*args, **kwargs)
  File "/tmp/_MEIqImArp/plugins/xkcd_comic/xkcd_comic.py", line 143, in update_function
    filename = comic_json['img'].split('/')[-1]
```

RishiKumarRay, 2022-01-12:
Can you guide me , how to solve this

txoof, 2022-01-12:
@RishiKumarRay Sure! It looks like the the JSON decode needs to be wrapped in a try/except block to trap for bad/missing JSON data.

It should be a pretty straightforward fix.

Do you have an EPD screen to test this against? If you don't you can do everything in Jupyter Notebook. There's some test code embedded in xkcd_comic.ipynb that you can use for testing and debugging. Check the last couple of cells.

Let me know if you need help getting up and going.

txoof, 2022-01-24:
resolved with commit 07c80f0916fcbfc204795d15b6114758950ab090

closed 2022-01-24 by txoof