txoof/epd_display #41: Support for Pimoroni Inky pHat

alemarengo, 2022-02-01:


txoof, 2022-02-01:
@alemarengo I'd love to integrate support for the Inky!

I don't have access to any Inky pHats, but I do have a pretty good idea how to move forward with this. Would you be willing to work with me in developing a solution? This would involve some refactoring of the underlying [epdlib](https://github.com/txoof/epdlib), but it was built with modularity in mind, so it should be fairly straightforward.

The [InkyPhat library](https://github.com/pimoroni/inky) looks pretty easy to use and would make it possible to add color images to PaperPi.

aaronr8684, 2022-03-08:
Hey @alemarengo, do you have some time to test the EPDLib changes to verify that the pHats are working as expected?

closed 2023-06-23 by txoof