txoof/epd_display #43: Paperpi doesn'

sprogus, 2022-02-02:
pi@raspberrypi:~ $ ./paperpi/dist/paperpi
09:22:53 paperpi:setup_display:325:CRITICAL - Error loading epd from configuration
09:22:53 paperpi:main:654:ERROR - config files used: [PosixPath('/tmp/_MEIh8nd8Z/config/paperpi.ini'), PosixPath('/home/pi/.config/com.txoof.paperpi/paperpi.ini')]
09:22:53 paperpi:do_exit:51:ERROR - failure caused exit: could not load epd module: epd2in13_V2 -- error: failed to load epd2in13_V2 with error: No module named 'numpy'

######################################################################
could not load epd module: epd2in13_V2 -- error: failed to load epd2in13_V2 with error: No module named 'numpy'
######################################################################

***Exiting***

txoof, 2022-02-02:
@sprogus 
PaperPi uses its own packaged python that's rolled into the executable. It doesn't search the system python.

This problem is related to the WaveShare modules trying to load unneeded modules. Let me see if I can fix this in a sane way and then roll up a new executable for you. 

In the meantime, can try using the `epd2in13` module instead? That might be forwards compatible with the V2 model.

txoof, 2022-02-02:
@sprogus Sorry for the multiple messages: the V3 module might work. It doesn't include the erroneous numpy import.

I'll see if I can fix this properly later.

txoof, 2022-02-02:
@sprogus 
Can you try the version found [here](https://github.com/txoof/epd_display/raw/I%2343_numpy_issue/paperpi_latest.tgz)? I think I have a reasonable way of patching this issue.

Thanks!

sprogus, 2022-02-02:
@txoof thanks for the reply. I'm now getting the below error

pi@raspberrypi:~ $ ./paperpi/dist/paperpi
12:14:12 paperpi:setup_display:325:CRITICAL - Error loading epd from configuration
12:14:12 paperpi:main:654:ERROR - config files used: [PosixPath('/tmp/_MEIo239h6/config/paperpi.ini'), PosixPath('/home/pi/.config/com.txoof.paperpi/paperpi.ini')]
12:14:12 paperpi:do_exit:51:ERROR - failure caused exit: could not load epd module: epd2in13_V2 -- error: failed to init display: init() missing 1 required positional argument: 'update'

######################################################################
could not load epd module: epd2in13_V2 -- error: failed to init display: init() missing 1 required positional argument: 'update'
######################################################################

***Exiting***

txoof, 2022-02-02:
@sprogus 
Doh. This is another of waveshare's non-standard [`init()` methods](https://github.com/txoof/epd_display/blob/81f76f380ac920a65f05d3b01357a4c3b932622f/paperpi/waveshare_epd/epd2in13_V2.py#L125). 

I've patched it up to default to `update=True`

Can you try again? https://github.com/txoof/epd_display/raw/I%2343_numpy_issue/paperpi_latest.tgz

If this works, I'll add it to my ugly patch script for future builds.

sprogus, 2022-02-02:
@txoof Thanks again. I've run the your patched version and now getting the below error

pi@raspberrypi:~ $ ./paperpi/dist/paperpi
22:15:39 paperpi:setup_display:325:CRITICAL - Error loading epd from configuration
22:15:39 paperpi:main:654:ERROR - config files used: [PosixPath('/tmp/_MEI9Qj1Nb/config/paperpi.ini'), PosixPath('/home/pi/.config/com.txoof.paperpi/paperpi.ini')]
22:15:39 paperpi:do_exit:51:ERROR - failure caused exit: could not load epd module: epd2in13_V2 -- error: failed to clear screen: Clear() missing 1 required positional argument: 'color'

######################################################################
could not load epd module: epd2in13_V2 -- error: failed to clear screen: Clear() missing 1 required positional argument: 'color'
######################################################################

***Exiting***

txoof, 2022-02-03:
@sprogus well, were getting closer and closer.

I have a few more ideas i can see through if you're willing to keep testing. 

Hopefully sometime today or tomorrow I'll try to patch this.

sprogus, 2022-02-03:
@txoof Yeah. More than happy to keep testing. Thanks for your effort so far.

txoof, 2022-02-03:
I'm contemplating a rewrite and migrating to the [Rob Webber's Omni EPD](https://github.com/robweber/omni-epd). It will take a lot of work on the backend to make this work, but it's probably worth it in the long run rather than dealing with all the hacky solutions I keep coming up with.

If we do get to a working solution, you'll probably want to try out some of the different layouts; the defaults don't look great on tiny screens.

txoof, 2022-02-03:
@sprogus 
Once again for the third time:  https://github.com/txoof/epd_display/raw/I%2343_numpy_issue/paperpi_latest.tgz

Keep me posted.

sprogus, 2022-02-03:
@txoof  We have success. Getting a display. I've just got to enable some plugins.

txoof, 2022-02-03:
Outstanding! If this works, I'll roll those changes into the main branch and publish an updated version.

If the existing layouts don't work for you, we can try to develop some that look better on a 2" display.

txoof, 2022-02-03:
@sprogus can you report back after you've tested this for a while and let me know if you run into any issues?

I'll let it run on my end with a 5.83" screen (unpatched) and see if I run into anything odd as well. 

Thanks for the testing support.

markbirss, 2022-02-20:
@txoof 

Hi

Im having this similar error with a epd2in13 (v1 display)

` wget https://github.com/txoof/epd_display/raw/master/paperpi_latest.tgz`

`11:40:07 paperpi:setup_display:325:CRITICAL - Error loading epd from configuration
11:40:07 paperpi:main:654:ERROR - config files used: [PosixPath('/tmp/_MEICL822T/config/paperpi.ini'), PosixPath('/home/pi/.config/com.txoof.paperpi/paperpi.ini')]
11:40:07 paperpi:do_exit:51:ERROR - failure caused exit: could not load epd module: epd2in13 -- error: failed to load epd2in13 with error: No module named 'numpy'

######################################################################
could not load epd module: epd2in13 -- error: failed to load epd2in13 with error: No module named 'numpy'
######################################################################

***Exiting***
`

txoof, 2022-02-20:
@markbirss The WaveShare code paperpi runs on is full of unnecessary imports. Try this patched version: https://github.com/txoof/epd_display/raw/I%2343_numpy_issue/paperpi_latest.tgz

It would be super helpful to hear if this works. If you don't run into any trouble, I'll merge these patches into the main branch. We're working on getting rid of the waveshare libraries altogether, but that's a ways off.

@sprogus if you have anything further to report, I would love to hear about it so I can merge these changes into the main branch.

sprogus, 2022-02-20:
Everything has been going well for me so far. I have only used the crypto
ticker though. I'm happy to test the other plugins if needed

markbirss, 2022-02-20:
@txoof 

I get this error ?

`pi@raspberrypi:~ $ ./paperpi/dist/paperpi
12:03:55 paperpi:setup_display:325:CRITICAL - Error loading epd from configuration
12:03:55 paperpi:main:654:ERROR - config files used: [PosixPath('/tmp/_MEItugbLx/config/paperpi.ini'), PosixPath('/home/pi/.config/com.txoof.paperpi/paperpi.ini')]
12:03:55 paperpi:do_exit:51:ERROR - failure caused exit: could not load epd module: epd2in13 -- error: failed to init display: init() missing 1 required positional argument: 'lut'

######################################################################
could not load epd module: epd2in13 -- error: failed to init display: init() missing 1 required positional argument: 'lut'
######################################################################

***Exiting***
`

`pi@raspberrypi:~ $ ./paperpi/dist/paperpi
12:02:36 paperpi:setup_display:325:CRITICAL - Error loading epd from configuration
12:02:36 paperpi:main:654:ERROR - config files used: [PosixPath('/tmp/_MEIuHhrJg/config/paperpi.ini'), PosixPath('/home/pi/.config/com.txoof.paperpi/paperpi.ini')]
12:02:36 paperpi:do_exit:51:ERROR - failure caused exit: could not load epd module: nano /home/pi/.config/com.txoof.paperpi/paperpi.ini -- error: unrecongized waveshare module: nano /home/pi/.config/com.txoof.paperpi/paperpi.ini

######################################################################
could not load epd module: nano /home/pi/.config/com.txoof.paperpi/paperpi.ini -- error: unrecongized waveshare module: nano /home/pi/.config/com.txoof.paperpi/paperpi.ini
######################################################################

***Exiting***
`

txoof, 2022-02-20:
@markbirss Thanks for the update. Let me see what I can do to patch that.

We're working on moving to OmniEPD to get rid of the waveshare code entirely. The sample code is a bit of a mess.

markbirss, 2022-02-20:
@txoof 

Thank you, my 9.7" ITE8951 is working good, just had small issue with those tiny jumpers for SPI mode, they seemed correctly set, but had to toggle them back and forth (maybe they were half way)

txoof, 2022-02-20:
Glad to hear that the 9.7" is working properly! I'd love to see any build pictures if you would be willing to share.

markbirss, 2022-02-20:
Oh ok, it looks similar to this, i just have not close up the case

[image]

This is running a hardware flow serial console over ESP32 based driver board
https://github.com/vroland/epdiy

https://cdn.hackaday.io/files/1681937195969312/term_demo_r.mp4
https://cdn.hackaday.io/files/1681937195969312/terminal_demo.mp4

txoof, 2022-02-20:
Wow. That's super cool. Is it "fast" enough to use as a monitor? I have a spare screen kicking around. It would be neat to use it as an interface for one of my many pi...

txoof, 2022-02-20:
@markbirss

Give this one a spin. It **should** fix the error you're seeing, but it may also generate other problems.  https://github.com/txoof/epd_display/raw/I%2343_numpy_issue/paperpi_latest.tgz

This screen supports partial refresh and the init() method expects a lut that matches either a full-refresh or a partial-refresh. I've hacked it so it  defaults to full refresh in all which matches most of the other displays.

Let me know how it works. If this is successful, I'll merge it into the master.

markbirss, 2022-02-20:
@txoof 

yes, they even working on a faster 16 grey color version

You must join the Slack group for the EPDIY Project

https://user-images.githubusercontent.com/22388007/154843582-26425fdc-3e71-42af-8b1a-7cf2fc48259e.mp4

It get this ?
`  271  wget -c https://github.com/txoof/epd_display/raw/I%2343_numpy_issue/paperpi_latest.tgz
  272  tar xvf paperpi_latest.tgz 
  273  history 
pi@raspberrypi:~ $ ./paperpi/dist/paperpi
Traceback (most recent call last):
  File "paperpi/paperpi.py", line 729, in <module>
  File "paperpi/paperpi.py", line 648, in main
  File "paperpi/paperpi.py", line 322, in setup_display
  File "epdlib/Screen.py", line 226, in __init__
  File "epdlib/Screen.py", line 38, in new_f
  File "epdlib/Screen.py", line 319, in epd
  File "epdlib/Screen.py", line 450, in _load_non_hd
  File "importlib/__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/tmp/_MEIxweJua/waveshare_epd/epd2in13.py", line 41, in <module>
    class EPD:
  File "/tmp/_MEIxweJua/waveshare_epd/epd2in13.py", line 99, in EPD
    def init(self, lut=self.lut_full_update):
NameError: name 'self' is not defined
[794] Failed to execute script 'paperpi' due to unhandled exception!
`

markbirss, 2022-02-20:
`271  wget -c https://github.com/txoof/epd_display/raw/I%2343_numpy_issue/paperpi_latest.tgz
  272  tar xvf paperpi_latest.tgz 
  273  history 
pi@raspberrypi:~ $ ./paperpi/dist/paperpi
Traceback (most recent call last):
  File "paperpi/paperpi.py", line 729, in <module>
  File "paperpi/paperpi.py", line 648, in main
  File "paperpi/paperpi.py", line 322, in setup_display
  File "epdlib/Screen.py", line 226, in __init__
  File "epdlib/Screen.py", line 38, in new_f
  File "epdlib/Screen.py", line 319, in epd
  File "epdlib/Screen.py", line 450, in _load_non_hd
  File "importlib/__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/tmp/_MEIxweJua/waveshare_epd/epd2in13.py", line 41, in <module>
    class EPD:
  File "/tmp/_MEIxweJua/waveshare_epd/epd2in13.py", line 99, in EPD
    def init(self, lut=self.lut_full_update):
NameError: name 'self' is not defined
[794] Failed to execute script 'paperpi' due to unhandled exception!`

markbirss, 2022-02-20:
271  wget -c https://github.com/txoof/epd_display/raw/I%2343_numpy_issue/paperpi_latest.tgz
  272  tar xvf paperpi_latest.tgz 
  273  history 
pi@raspberrypi:~ $ ./paperpi/dist/paperpi
Traceback (most recent call last):
  File "paperpi/paperpi.py", line 729, in <module>
  File "paperpi/paperpi.py", line 648, in main
  File "paperpi/paperpi.py", line 322, in setup_display
  File "epdlib/Screen.py", line 226, in __init__
  File "epdlib/Screen.py", line 38, in new_f
  File "epdlib/Screen.py", line 319, in epd
  File "epdlib/Screen.py", line 450, in _load_non_hd
  File "importlib/__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/tmp/_MEIxweJua/waveshare_epd/epd2in13.py", line 41, in <module>
    class EPD:
  File "/tmp/_MEIxweJua/waveshare_epd/epd2in13.py", line 99, in EPD
    def init(self, lut=self.lut_full_update):
NameError: name 'self' is not defined
[794] Failed to execute script 'paperpi' due to unhandled exception!

markbirss, 2022-02-20:
This is the ITE8951 9.7inch im running your epd_display on

[image]

txoof, 2022-02-20:
Uuuh. I should have read the waveshare code more closely. There's no easy fix, but there is a kludge!

I've tested this as much as I can without an actual 2in display. Can you try again?

https://github.com/txoof/epd_display/raw/I%2343_numpy_issue/paperpi_latest.tgz

markbirss, 2022-02-20:
pi@raspberrypi:~ $ ./paperpi/./dist/paperpi
17:42:38 Block:padding:315:WARNING - padding value is greater >= 1/2 of area dimension 12, no image will be displayed: 10
17:42:38 Block:padding:322:WARNING - the padded area available may be too small to display any content: Area: (250, 12), Padded Area: [230, -8]
17:42:38 Block:font_size:795:WARNING - no font size set, using {font_size}

txoof, 2022-02-20:
Good news! I'll work on merging this into the main branch soon.

Try setting the `rotation` param in the config file. That screen is probably better suited for a landscape display due to the way the text layout is handled.

Each plugin has multiple layouts. Some of the layouts are designed to have a lower information density so they display better on small screens. Check the [provided layouts](https://github.com/txoof/epd_display/blob/master/paperpi/plugins/xkcd_comic/README.md#provided-layouts) section of each plugin to get an idea of what they look like.

If you find that the plugins look lousy on this screen, we can work on creating better layouts that work on tiny screens. 

Right now it's a giant pain in the butt to update the layouts due to the way we're distributing the blob. In the near future we'll be moving to a much more flexible system that allows easy editing. You can get an early preview [here](https://github.com/txoof/PaperPi/tree/manage_modules).

The install instructions are [here](https://github.com/txoof/PaperPi/tree/manage_modules#install--setup).

markbirss, 2022-02-20:
Ok, cool, thank you

txoof, 2022-02-21:
@sprogus, @markbirss Changes merged into master.

closed 2022-02-21 by txoof