txoof/epd_display #48: Create_devel_venv.sh failed

aaronr8684, 2022-02-05:
See the output below. Issue was fixed by manually running `sudo apt-get install libcairo2-dev libjpeg-dev libgif-dev` and then running the shell script again. I believe this should be a quick fix in the "cd_apt_packages" file, but, I'm not sure. If it is just a simple add to the CORE var, I'm happy to make the change and submit the PR.

```An error occurred while installing pycairo! Will try again.
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 22/22 — 00:02:22
Installing initially failed dependencies...
[pipenv.exceptions.InstallError]: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[pipenv.exceptions.InstallError]: Collecting pycairo
[pipenv.exceptions.InstallError]:   Using cached pycairo-1.20.1.tar.gz (344 kB)
[pipenv.exceptions.InstallError]:   Preparing metadata (setup.py): started
[pipenv.exceptions.InstallError]:   Preparing metadata (setup.py): finished with status 'done'
[pipenv.exceptions.InstallError]: Building wheels for collected packages: pycairo
[pipenv.exceptions.InstallError]:   Building wheel for pycairo (setup.py): started
[pipenv.exceptions.InstallError]:   Building wheel for pycairo (setup.py): finished with status 'error'
[pipenv.exceptions.InstallError]:   Running setup.py clean for pycairo
[pipenv.exceptions.InstallError]: Failed to build pycairo
[pipenv.exceptions.InstallError]: Installing collected packages: pycairo
[pipenv.exceptions.InstallError]:     Running setup.py install for pycairo: started
[pipenv.exceptions.InstallError]:     Running setup.py install for pycairo: finished with status 'error'
[pipenv.exceptions.InstallError]: ERROR: Command errored out with exit status 1:
[pipenv.exceptions.InstallError]:    command: /home/pi/.local/share/virtualenvs/epd_display-TrMBV6hi/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h9ewg1fo/pycairo_6e9b912de55f42a5b3452cd78a143e61/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h9ewg1fo/pycairo_6e9b912de55f42a5b3452cd78a143e61/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-adj3i4qg
[pipenv.exceptions.InstallError]:        cwd: /tmp/pip-install-h9ewg1fo/pycairo_6e9b912de55f42a5b3452cd78a143e61/
[pipenv.exceptions.InstallError]:   Complete output (17 lines):
[pipenv.exceptions.InstallError]:   running bdist_wheel
[pipenv.exceptions.InstallError]:   running build
[pipenv.exceptions.InstallError]:   running build_py
[pipenv.exceptions.InstallError]:   creating build
[pipenv.exceptions.InstallError]:   creating build/lib.linux-aarch64-3.9
[pipenv.exceptions.InstallError]:   creating build/lib.linux-aarch64-3.9/cairo
[pipenv.exceptions.InstallError]:   copying cairo/__init__.py -> build/lib.linux-aarch64-3.9/cairo
[pipenv.exceptions.InstallError]:   copying cairo/__init__.pyi -> build/lib.linux-aarch64-3.9/cairo
[pipenv.exceptions.InstallError]:   copying cairo/py.typed -> build/lib.linux-aarch64-3.9/cairo
[pipenv.exceptions.InstallError]:   warning: build_py: byte-compiling is disabled, skipping.
[pipenv.exceptions.InstallError]:   
[pipenv.exceptions.InstallError]:   running build_ext
[pipenv.exceptions.InstallError]:   Package cairo was not found in the pkg-config search path.
[pipenv.exceptions.InstallError]:   Perhaps you should add the directory containing `cairo.pc'
[pipenv.exceptions.InstallError]:   to the PKG_CONFIG_PATH environment variable
[pipenv.exceptions.InstallError]:   No package 'cairo' found
[pipenv.exceptions.InstallError]:   Command '['pkg-config', '--print-errors', '--exists', 'cairo >= 1.15.10']' returned non-zero exit status 1.
[pipenv.exceptions.InstallError]:   ----------------------------------------
[pipenv.exceptions.InstallError]:   ERROR: Failed building wheel for pycairo
[pipenv.exceptions.InstallError]:     ERROR: Command errored out with exit status 1:
[pipenv.exceptions.InstallError]:      command: /home/pi/.local/share/virtualenvs/epd_display-TrMBV6hi/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h9ewg1fo/pycairo_6e9b912de55f42a5b3452cd78a143e61/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h9ewg1fo/pycairo_6e9b912de55f42a5b3452cd78a143e61/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-qc0zxiqm/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.local/share/virtualenvs/epd_display-TrMBV6hi/include/site/python3.9/pycairo
[pipenv.exceptions.InstallError]:          cwd: /tmp/pip-install-h9ewg1fo/pycairo_6e9b912de55f42a5b3452cd78a143e61/
[pipenv.exceptions.InstallError]:     Complete output (19 lines):
[pipenv.exceptions.InstallError]:     running install
[pipenv.exceptions.InstallError]:     /home/pi/.local/share/virtualenvs/epd_display-TrMBV6hi/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
[pipenv.exceptions.InstallError]:       warnings.warn(
[pipenv.exceptions.InstallError]:     running build
[pipenv.exceptions.InstallError]:     running build_py
[pipenv.exceptions.InstallError]:     creating build
[pipenv.exceptions.InstallError]:     creating build/lib.linux-aarch64-3.9
[pipenv.exceptions.InstallError]:     creating build/lib.linux-aarch64-3.9/cairo
[pipenv.exceptions.InstallError]:     copying cairo/__init__.py -> build/lib.linux-aarch64-3.9/cairo
[pipenv.exceptions.InstallError]:     copying cairo/__init__.pyi -> build/lib.linux-aarch64-3.9/cairo
[pipenv.exceptions.InstallError]:     copying cairo/py.typed -> build/lib.linux-aarch64-3.9/cairo
[pipenv.exceptions.InstallError]:     warning: build_py: byte-compiling is disabled, skipping.
[pipenv.exceptions.InstallError]:     
[pipenv.exceptions.InstallError]:     running build_ext
[pipenv.exceptions.InstallError]:     Package cairo was not found in the pkg-config search path.
[pipenv.exceptions.InstallError]:     Perhaps you should add the directory containing `cairo.pc'
[pipenv.exceptions.InstallError]:     to the PKG_CONFIG_PATH environment variable
[pipenv.exceptions.InstallError]:     No package 'cairo' found
[pipenv.exceptions.InstallError]:     Command '['pkg-config', '--print-errors', '--exists', 'cairo >= 1.15.10']' returned non-zero exit status 1.
[pipenv.exceptions.InstallError]:     ----------------------------------------
[pipenv.exceptions.InstallError]: ERROR: Command errored out with exit status 1: /home/pi/.local/share/virtualenvs/epd_display-TrMBV6hi/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h9ewg1fo/pycairo_6e9b912de55f42a5b3452cd78a143e61/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h9ewg1fo/pycairo_6e9b912de55f42a5b3452cd78a143e61/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-qc0zxiqm/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.local/share/virtualenvs/epd_display-TrMBV6hi/include/site/python3.9/pycairo Check the logs for full command output.
ERROR: Couldn't install package: pycairo
 Package installation failed...
  ☤  ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 0/1 — 00:00:10
exiting script ./create_devel_venv.sh
reason: failed to install modules
```

txoof, 2022-02-05:
What version of raspberry pi OS are you running? I recently tested this on a clean, vanilla 32 Bit Buster and didn't run into any trouble. I'd like to chase this down to the root cause if I can.

I'm not super happy with the web of scripts that live in the root of this project. For the moment, it's probably worth at least patching this up with a solution if you have one.

aaronr8684, 2022-02-05:
I just did a fresh install of Raspbian OS (64-bit) downloaded from their site using Raspberry Pi Imager.

Output of /etc/os-release
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

aaronr8684, 2022-02-28:
This is also fixed with the v0.3.0 of [PaperPi](https://github.com/txoof/paperpi)

closed 2022-02-28 by aaronr8684