txoof/epd_display #54: tar file corupt

ejmorgan55, 2022-02-08:
pi@raspberrypi:~ $ tar xvzf paperpi_latest.tgz
paperpi/./dist/paperpi

gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
cloned from your site due to tar not workintg. cant launch paperpi. not working.

aaronr8684, 2022-02-08:
@ejmorgan55 Did you try to download the file again? I just downloaded this an hour ago and it worked for me.

ejmorgan55, 2022-02-08:
I did; also, when I try to execute ./paperpi/dist/paperpi I get the error cannot open pyinstaller archive from executable?

ejmorgan55, 2022-02-09:
OK; after sever tries I was able to load the program again and it is working. However, I am getting a warning block init 183::warning - fill and background are identical and this likely result in no visible image? I have the phases of the moon; basid clock; and quotes selected. Appears phases of the moon is not work as I can see the clock and quotes?

txoof, 2022-02-09:
@ejmorgan55 
That warning can be safely ignored. It's probably just an oversight in the layout. 

I have a suspicion that I know what's going on with the tar. 

Can you share everything you see when it throws errors? 

Can you also share what you see when you type `lsb_release -a`?

ejmorgan55, 2022-02-09:
When asked for the lsb module it states none are available; raspbian gnu/linux 11 (bullseye). The error concerning warning block init 183::warning - fill and background are identical and this likely result in no visible image happens when you  place False for the splash screen; and indeed, all you get is a white screen if you do so. For the tar; I presume it may have been a download issue. As indicated, I downloaded it 3 times; on the 3d try, it worked without problems. I have the system up and operating now and it seems to work without error. Thanks for the help.

txoof, 2022-02-09:
@ejmorgan55 
thanks for the report. That's really helpful. I think the epdlib needs some work on the warning messages. That message is pretty extreme for what it means.

FYI, github supports markdown. You can surround log output, terminal copy/paste and code blocks by using the <> button. It adds "`" around the snip. If you have big code blocks, you can paste those by using three back-ticks. 

It helps make things a little more readable.

closed 2022-02-09 by txoof