txoof/epd_display #57: Add partial refresh capability

winstonma, 2022-02-16:
There is a hack that allows the [partial refresh happen](https://youtu.be/MsbiO8EAsGw), simply look at [15:12](https://youtu.be/MsbiO8EAsGw?t=912) and they just keep the last line of `WW`. Please check the old and the new table in 15:12 for the details.

I tested it and this method works on my 7.5" version 2 eink screen.

txoof, 2022-02-16:
Thanks for the suggestion. Would you submit a PR with the changes? 

We are also looking to move to the Omni EPD library rather than using the Waveshare code. Do you have any interest in helping bout with that? 

@aaronr8684 is working on merging in those changes.

winstonma, 2022-02-16:
I think this is the core issue of the problem.

For example

- 7.5" V2
   The [epd7in5_V2.py](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd7in5_V2.py) uses `LUT_WW_7IN5_V2`
- 4.2"
   The [epd4in2.py](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd4in2.py) uses `lut_ww`

I am not sure how to generalize this. Currently I have a [repo](https://github.com/winstonma/RPi-MagicMirror-Waveshare) that uses the chrome headless browser to access MagicMirror server, and then display the screenshot onto the eink device. I have a mod version running locally that support partial refresh on my 7.5" V2 device but I am not sure how to generalize that to all eink device that doesn't support partial refresh.

txoof, 2022-02-16:
@winstonma 
>  I am not sure how to generalize that to all eink device that doesn't support partial refresh.
That's pretty much the problem we ran into as well; rather than try to sort out the WaveShare code, I decided to just roll with the limitations for now.

OmniEPD resolves some of this, but I haven't completely explored it yet. There's a plan to move from the WaveShare libraries to Omni EPD so we can offer 3 and 7 color support. If partial refresh is supported for the other displays, we'll roll that in as well. 

All of that depends on making changes to the `epdlib` that backends this project. Aaronr has been working on getting some changes merged into Omni EPD library so we can then work those into `epdlib`. Ideally we will abandon WaveShare all together and just use the IT8951 libraries from Greg Meyer and Omni EPD in the future. 

We'd love to have your input if you have ideas if you're interested in helping out.

aaronr8684, 2022-02-16:
@winstonma Yea, unfortunately, it's not officially supported by Waveshare for the 7.5" screens. So, short of modifying the driver files directly (like Applied Science did), it's not something that is easily added for a "library" across all the screens. The OmniEPD library also doesn't support partial refresh yet, even on the displays that have official support for it (e.g. 2.13", 3.7", 6", etc.). I could see the official methods being implemented eventually, but right now, my priority is getting color/greyscale support added to this project. 

There is also the possibility of unofficial code breaking or at least decreasing the life of the screens. Since the aim of this library is to keep the user from needing to worry about the low-level, I'm not sure if adding unofficial features would be good for the long term support, especially as we move away from using the Waveshare drivers directly. If you have any ideas or solutions, I'm open to exploring them.

If you want to use this library but make use your own solution for partial update support, I would look at the solution I came up with for the 3.7" screen in issue #50 which involves "overloading" the methods of the original driver file while using the rest of the driver as designed. If you have any questions with that solution, let me know. I just don't see bringing in partial support for the screens that don't officially support it anytime soon (if ever).

Hope that helps!

winstonma, 2022-02-17:
@aaronr8684 Agree the partial refresh is a hack and it is not as good as the native partial refresh other screen provided. And that's the reason `hacking` is needed.

Yeah I am not sure what is best. At the same time I think it would be best if the epd_display could be split into two separate project - one server side and one client side. And the client side would display the server side content. And the client side focus on the display of content (eink) and the server would be all the content.

Thanks a lot

txoof, 2022-02-17:
@winstonma That's more-or-less how things are set up. PaperPi handles what to display and when while `epdlib` manages talking to the display and formatting the content to fit on whatever resolution screen is configured.

This means that most of the changes need to happen at the `epdlib` side. This is why we're looking to move to Omni EPD for the hardware interface. The most sustainable route for getting partial refreshes included would probably be to try and work those changes into Omni EPD. 

If partial refresh is available in Omni EPD, it should be fairly easy to add those features into PaperPi. It already supports partial refresh for HD panels, so it would just be a matter of changing some of the configuration options to allow that method to be used on any screen that supports partial refresh.

closed 2023-06-23 by txoof