txoof/epd_display #58: Paperpi freezing

ejmorgan55, 2022-02-23:
Hi; I have downloaded paperpi on a raspberry pi 3 using the current outlook rendition. I am operating a waveshare 4.2 screen. Download was fine and set up on Damon. I am running the word clock and phases of the moon. Problem is that the system starts, and works fine for about 12 hours. It then simply freezes; and does not function. I thought it might be a bad download; so I performed a new download with a new SD card. Same problem. I then changed to a different Raspberry Pi 3b that I had sitting around. Same problem. As it starts and functions for 12 hours; I doubt it has anything to do with the wiring. Any ideas? Looks great when operating; however, having to restart it every 12 hours is a non player. Please let me know if there is something else I can do. Thanks.

txoof, 2022-02-23:
Interesting problem I haven't seen before.

Does the entire pi freeze up, or just PaperPi? What I mean is can you ssh in or still interact via keyboard/monitor?

What version of RasPiOS are you using? Run `$ uname -a`.

Can you do the following:
- edit `/etc/defaults/paperpi.ini` and change logging to "DEBUG"
- open a terminal (preferably over ssh with tmux) and run `$ tail -f /var/log/daemon.log`

You should see lots and lots of output from PaperPi in the Daemon log output. After it freezes, grab any error messages or anything that looks relevant and share it here.

I'll do the same and let it run today to see if I get a similar problem.

ejmorgan55, 2022-02-23:
As requested above:

Linux raspberrypi 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/Linux

log level changed to debug

restarted as above; however, paperpi does not restart and the screen remains frozen. Following was the result from the log:
 tail -f /var/log/daemon.log
Feb 23 11:02:53 raspberrypi paperpi[1225]: 11:02:53 paperpi:main:645:DEBUG - configuration:
Feb 23 11:02:53 raspberrypi paperpi[1225]: {'main': {'display_type': 'epd4in2b_V2', 'vcom': 0.0, 'max_refresh': 4, 'log_level': 'DEBUG', 'splash': 1, 'rotation': 0, 'daemon': True}, 'xPlugin: decimal binary clock': {'layout': 'layout', 'plugin': 'dec_bin_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, 'xPlugin: Weather Adis Ababa': {'layout': 'layout', 'plugin': 'met_no', 'refresh_rate': 300, 'min_display_time': 40, 'max_priority': 2, 'location_name': 'Adis Ababa', 'lat': 9.0, 'lon': 38.75, 'email': 'you@host.diamond'}, 'xPlugin: Crypto Bitcoin v USD': {'plugin': 'crypto', 'layout': 'layout', 'fiat': 'usd', 'coin': 'bitcoin', 'days': 14, 'interval': 'hourly', 'rss_feed': 'https://bitcoinmagazine.com/.rss/full/', 'min_display_time': 30, 'refresh_rate': 300, 'max_priority': 2}, 'xPlugin: Moon Phase': {'layout': 'layout', 'plugin': 'moon_phase', 'min_display_time': 30, 'max_priority': 2, 'email': 'you@host.diamond', 'location_name': 'Europe/Amsterdam', 'lat': 52.3, 'lon': 4.9}, 'xPlugin: A Demo Plugin': {'layout': 'layout', 'plugin': 'demo_plugin', 'min_display_time': 30, 'max_priority': 1, 'your_name': 'Slartybartfast', 'your_color': 'chartreuse'}, 'xPlugin: Basic Clock': {'layout': 'layout', 'plugin': 'basic_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, 'xPlugin: XKCD Comic Plugin': {'layout': 'layout', 'plugin': 'xkcd_comic', 'refresh_rate': 1200, 'min_display_time': 120, 'max_x': 800, 'max_y': 600, 'resize': 0, 'max_retries': 10, 'max_priority': 2}, 'xPlugin: Home Assistant': {'layout': 'layout', 'plugin': 'home_assistant', 'min_display_time': 30, 'max_priority': 1, 'home_assistant_basepath': 'https://IP:8123', 'home_assistant_token': '', 'entity1_name': 'Friendly name (ex Indoor)', 'entity1_id': 'home assistant sensor id', 'entity2_name': 'Friendly name (ex Outdoor)', 'entity2_id': 'home assistant sensor id', 'entity3_name': 'Friendly name (ex Outdoor)', 'entity3_id': 'home assistant sensor id', 'entity4_name': 'Friendly name (ex Outdoor)', 'entity5_id': 'home assistant sensor id', 'media_id': 'media player id'}, 'xPlugin: New Yorker Comic': {'layout': 'layout', 'plugin': 'newyorker', 'day_range': 5, 'refresh_rate': 120, 'min_display_time': 60, 'max_priority': 2}, 'xPlugin: Librespot': {'layout': 'layout', 'plugin': 'librespot_client', 'refresh_rate': 10, 'max_priority': 0, 'min_display_time': 15, 'player_name': 'SpoCon-Spotify', 'idle_timeout': 10}, 'xPlugin: Reddit Quotes': {'layout': 'layout', 'plugin': 'reddit_quote', 'refresh_rate': 100, 'min_display_time': 50, 'max_priority': 2, 'max_length': 144}, 'xPlugin: LMS - Your Player Name': {'layout': 'layout', 'plugin': 'lms_client', 'player_name': 'Your Player Name', 'refresh_rate': 5, 'min_display_time': 30, 'max_priority': 0, 'idle_timeout': 15}, 'xPlugin: Word Clock': {'layout': 'layout', 'plugin': 'word_clock', 'refresh_rate': 125, 'min_display_time': 255, 'max_priority': 2}, 'Plugin: Moon Phase': {'layout': 'layout', 'plugin': 'moon_phase', 'min_display_time': 30, 'max_priority': 2, 'email': 'ejmorgan@msn.com', 'location_name': 'America/Phoenix', 'lat': 31.54, 'lon': 110.27}, 'Plugin: Word Clock': {'layout': 'layout', 'plugin': 'word_clock', 'refresh_rate': 125, 'min_display_time': 255, 'max_priority': 2}, '__cmd_line': {'list_plugins': False}}
Feb 23 11:02:53 raspberrypi paperpi[1225]: 11:02:53 Screen:_load_non_hd:442:DEBUG - configuring waveshare_epd.epd4in2b_V2
Feb 23 11:02:53 raspberrypi paperpi[1225]: 11:02:53 Screen:_load_non_hd:472:DEBUG - args_spec: ['self', 'imageblack', 'imagered']
Feb 23 11:02:53 raspberrypi paperpi[1225]: 11:02:53 Screen:epd:332:DEBUG - epd configuration {'epd': <waveshare_epd.epd4in2b_V2.EPD object at 0x7528d130>, 'resolution': [300, 400], 'clear_args': {}, 'one_bit_display': False, 'constants': None, 'mode': '1'}
Feb 23 11:02:53 raspberrypi paperpi[1225]: 11:02:53 epd4in2b_V2:getbuffer:97:DEBUG - Horizontal
Feb 23 11:02:53 raspberrypi paperpi[1225]: 11:02:53 Screen:rotation:379:DEBUG - rotation=0, resolution=[400, 300]
Feb 23 11:02:53 raspberrypi paperpi[1225]: 11:02:53 Screen:wrapper:246:DEBUG - initing display
Feb 23 11:02:53 raspberrypi paperpi[1225]: 11:02:53 Screen:wrapper:249:DEBUG - Non HD display
Feb 23 11:02:54 raspberrypi paperpi[1225]: 11:02:54 epd4in2b_V2:ReadBusy:68:DEBUG - e-Paper busy

txoof, 2022-02-23:
This appears to be just the log entries from the initial startup. Everything looks pretty OK up to this point.  Does it freeze immediately after `11:02:54` or does it take 12 hours? It would be super helpful to see the log entries from around the time it freezes.

If you can capture just the `/var/log/daemon.log` lines from around the time it freezes, that would be super helpful.  

The next time it freezes try this:

`sudo systemctl stop paperpi-daemon; sudo systemctl start paperpi-daemon` 

Take a look in the logs and see if anything interesting happens around the time you stop it and start it.

I've got a 3B running Bullseye 32 bit `Linux raspberrypi 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/` same as with an epd5in83 running here on my desk; it's been going for right around 13 hours . I'll keep it going for a few days to see if I can reproduce this problem.

ejmorgan55, 2022-02-23:
Froze again; here is the log:
pi@raspberrypi:~/paperpi $ tail -f /var/log/daemon.log
Feb 23 11:37:59 raspberrypi dhcpcd[533]: wlan0: carrier lost
Feb 23 11:38:00 raspberrypi dhcpcd[533]: wlan0: deleting address fe80::8073:7b84:7df6:38c0
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Withdrawing address record for fe80::8073:7b84:7df6:38c0 on wlan0.
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::8073:7b84:7df6:38c0.
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Interface wlan0.IPv6 no longer relevant for mDNS.
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Withdrawing address record for 192.168.86.22 on wlan0.
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.86.22.
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Interface wlan0.IPv4 no longer relevant for mDNS.
Feb 23 11:38:00 raspberrypi dhcpcd[533]: wlan0: deleting route to 192.168.86.0/24
Feb 23 11:38:00 raspberrypi dhcpcd[533]: wlan0: deleting default route via 192.168.86.1
pi@raspberrypi:~/paperpi $ tail -f /var/log/daemon.log
Feb 23 11:37:59 raspberrypi dhcpcd[533]: wlan0: carrier lost
Feb 23 11:38:00 raspberrypi dhcpcd[533]: wlan0: deleting address fe80::8073:7b84:7df6:38c0
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Withdrawing address record for fe80::8073:7b84:7df6:38c0 on wlan0.
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::8073:7b84:7df6:38c0.
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Interface wlan0.IPv6 no longer relevant for mDNS.
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Withdrawing address record for 192.168.86.22 on wlan0.
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.86.22.
Feb 23 11:38:00 raspberrypi avahi-daemon[451]: Interface wlan0.IPv4 no longer relevant for mDNS.
Feb 23 11:38:00 raspberrypi dhcpcd[533]: wlan0: deleting route to 192.168.86.0/24
Feb 23 11:38:00 raspberrypi dhcpcd[533]: wlan0: deleting default route via 192.168.86.1

ejmorgan55, 2022-02-23:
Tried sudo systemctl stop paperpi-daemon; sudo systemctl start paperpi-daemon. No restart.

ejmorgan55, 2022-02-23:
After trying to restart:
pi@raspberrypi:~/paperpi $ tail -f /var/log/daemon.log
Feb 23 11:42:48 raspberrypi paperpi[1451]: 11:42:48 paperpi:main:645:DEBUG - configuration:
Feb 23 11:42:48 raspberrypi paperpi[1451]: {'main': {'display_type': 'epd4in2b_V2', 'vcom': 0.0, 'max_refresh': 4, 'log_level': 'DEBUG', 'splash': 1, 'rotation': 0, 'daemon': True}, 'xPlugin: decimal binary clock': {'layout': 'layout', 'plugin': 'dec_bin_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, 'xPlugin: Weather Adis Ababa': {'layout': 'layout', 'plugin': 'met_no', 'refresh_rate': 300, 'min_display_time': 40, 'max_priority': 2, 'location_name': 'Adis Ababa', 'lat': 9.0, 'lon': 38.75, 'email': 'you@host.diamond'}, 'xPlugin: Crypto Bitcoin v USD': {'plugin': 'crypto', 'layout': 'layout', 'fiat': 'usd', 'coin': 'bitcoin', 'days': 14, 'interval': 'hourly', 'rss_feed': 'https://bitcoinmagazine.com/.rss/full/', 'min_display_time': 30, 'refresh_rate': 300, 'max_priority': 2}, 'xPlugin: Moon Phase': {'layout': 'layout', 'plugin': 'moon_phase', 'min_display_time': 30, 'max_priority': 2, 'email': 'you@host.diamond', 'location_name': 'Europe/Amsterdam', 'lat': 52.3, 'lon': 4.9}, 'xPlugin: A Demo Plugin': {'layout': 'layout', 'plugin': 'demo_plugin', 'min_display_time': 30, 'max_priority': 1, 'your_name': 'Slartybartfast', 'your_color': 'chartreuse'}, 'xPlugin: Basic Clock': {'layout': 'layout', 'plugin': 'basic_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, 'xPlugin: XKCD Comic Plugin': {'layout': 'layout', 'plugin': 'xkcd_comic', 'refresh_rate': 1200, 'min_display_time': 120, 'max_x': 800, 'max_y': 600, 'resize': 0, 'max_retries': 10, 'max_priority': 2}, 'xPlugin: Home Assistant': {'layout': 'layout', 'plugin': 'home_assistant', 'min_display_time': 30, 'max_priority': 1, 'home_assistant_basepath': 'https://IP:8123', 'home_assistant_token': '', 'entity1_name': 'Friendly name (ex Indoor)', 'entity1_id': 'home assistant sensor id', 'entity2_name': 'Friendly name (ex Outdoor)', 'entity2_id': 'home assistant sensor id', 'entity3_name': 'Friendly name (ex Outdoor)', 'entity3_id': 'home assistant sensor id', 'entity4_name': 'Friendly name (ex Outdoor)', 'entity5_id': 'home assistant sensor id', 'media_id': 'media player id'}, 'xPlugin: New Yorker Comic': {'layout': 'layout', 'plugin': 'newyorker', 'day_range': 5, 'refresh_rate': 120, 'min_display_time': 60, 'max_priority': 2}, 'xPlugin: Librespot': {'layout': 'layout', 'plugin': 'librespot_client', 'refresh_rate': 10, 'max_priority': 0, 'min_display_time': 15, 'player_name': 'SpoCon-Spotify', 'idle_timeout': 10}, 'xPlugin: Reddit Quotes': {'layout': 'layout', 'plugin': 'reddit_quote', 'refresh_rate': 100, 'min_display_time': 50, 'max_priority': 2, 'max_length': 144}, 'xPlugin: LMS - Your Player Name': {'layout': 'layout', 'plugin': 'lms_client', 'player_name': 'Your Player Name', 'refresh_rate': 5, 'min_display_time': 30, 'max_priority': 0, 'idle_timeout': 15}, 'xPlugin: Word Clock': {'layout': 'layout', 'plugin': 'word_clock', 'refresh_rate': 125, 'min_display_time': 255, 'max_priority': 2}, 'Plugin: Moon Phase': {'layout': 'layout', 'plugin': 'moon_phase', 'min_display_time': 30, 'max_priority': 2, 'email': 'ejmorgan@msn.com', 'location_name': 'America/Phoenix', 'lat': 31.54, 'lon': 110.27}, 'Plugin: Word Clock': {'layout': 'layout', 'plugin': 'word_clock', 'refresh_rate': 125, 'min_display_time': 255, 'max_priority': 2}, '__cmd_line': {'list_plugins': False}}
Feb 23 11:42:48 raspberrypi paperpi[1451]: 11:42:48 Screen:_load_non_hd:442:DEBUG - configuring waveshare_epd.epd4in2b_V2
Feb 23 11:42:48 raspberrypi paperpi[1451]: 11:42:48 Screen:_load_non_hd:472:DEBUG - args_spec: ['self', 'imageblack', 'imagered']
Feb 23 11:42:48 raspberrypi paperpi[1451]: 11:42:48 Screen:epd:332:DEBUG - epd configuration {'epd': <waveshare_epd.epd4in2b_V2.EPD object at 0x752ae110>, 'resolution': [300, 400], 'clear_args': {}, 'one_bit_display': False, 'constants': None, 'mode': '1'}
Feb 23 11:42:48 raspberrypi paperpi[1451]: 11:42:48 epd4in2b_V2:getbuffer:97:DEBUG - Horizontal
Feb 23 11:42:49 raspberrypi paperpi[1451]: 11:42:49 Screen:rotation:379:DEBUG - rotation=0, resolution=[400, 300]
Feb 23 11:42:49 raspberrypi paperpi[1451]: 11:42:49 Screen:wrapper:246:DEBUG - initing display
Feb 23 11:42:49 raspberrypi paperpi[1451]: 11:42:49 Screen:wrapper:249:DEBUG - Non HD display
Feb 23 11:42:49 raspberrypi paperpi[1451]: 11:42:49 epd4in2b_V2:ReadBusy:68:DEBUG - e-Paper busy
pi@raspberrypi:~/paperpi $ tail -f /var/log/daemon.log
Feb 23 11:42:48 raspberrypi paperpi[1451]: 11:42:48 paperpi:main:645:DEBUG - configuration:
Feb 23 11:42:48 raspberrypi paperpi[1451]: {'main': {'display_type': 'epd4in2b_V2', 'vcom': 0.0, 'max_refresh': 4, 'log_level': 'DEBUG', 'splash': 1, 'rotation': 0, 'daemon': True}, 'xPlugin: decimal binary clock': {'layout': 'layout', 'plugin': 'dec_bin_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, 'xPlugin: Weather Adis Ababa': {'layout': 'layout', 'plugin': 'met_no', 'refresh_rate': 300, 'min_display_time': 40, 'max_priority': 2, 'location_name': 'Adis Ababa', 'lat': 9.0, 'lon': 38.75, 'email': 'you@host.diamond'}, 'xPlugin: Crypto Bitcoin v USD': {'plugin': 'crypto', 'layout': 'layout', 'fiat': 'usd', 'coin': 'bitcoin', 'days': 14, 'interval': 'hourly', 'rss_feed': 'https://bitcoinmagazine.com/.rss/full/', 'min_display_time': 30, 'refresh_rate': 300, 'max_priority': 2}, 'xPlugin: Moon Phase': {'layout': 'layout', 'plugin':  [... cut ...]
Feb 23 11:42:48 raspberrypi paperpi[1451]: 11:42:48 Screen:_load_non_hd:472:DEBUG - args_spec: ['self', 'imageblack', 'imagered']
Feb 23 11:42:48 raspberrypi paperpi[1451]: 11:42:48 Screen:epd:332:DEBUG - epd configuration {'epd': <waveshare_epd.epd4in2b_V2.EPD object at 0x752ae110>, 'resolution': [300, 400], 'clear_args': {}, 'one_bit_display': False, 'constants': None, 'mode': '1'}
Feb 23 11:42:48 raspberrypi paperpi[1451]: 11:42:48 epd4in2b_V2:getbuffer:97:DEBUG - Horizontal
Feb 23 11:42:49 raspberrypi pape

txoof, 2022-02-23:
@ejmorgan55 
OK. Let's try something different. What I really want to see is just the stuff related to paperpi and maybe some other stuff that's going on around it.

try this:  `journalctl --since HH:MM --until "now" paperpi.log` **see the note below**

Then attach the log to a new comment so I can take a look. You might want to review the log to make sure there's nothing you don't want the world to see in there.
## NOTE
**You need to substitute a time for HH:MM in 24 hour time. e.g. 09:59 or 19:44** Try to pick a time that's close to when it failed.

txoof, 2022-02-23:
Sorry, should read: `journalctl --since HH:MM --until "now" > paperpi.log`

ejmorgan55, 2022-02-23:
OK; had to wait for it to freeze again:

pi@raspberrypi:~ $ tail -f /var/log/daemon.log
Feb 23 14:37:47 raspberrypi paperpi[713]: 14:37:47 paperpi:update_plugins:466:DEBUG - IMAGE STRING: <PIL.Image.Image image mode=L size=400x300 at 0x752393F0>
Feb 23 14:37:47 raspberrypi paperpi[713]: 14:37:47 paperpi:update_loop:513:INFO - display_time elapsed, cycling to next active plugin
Feb 23 14:37:47 raspberrypi paperpi[713]: 14:37:47 paperpi:update_loop:522:DEBUG - searching for next active plugin
Feb 23 14:37:47 raspberrypi paperpi[713]: 14:37:47 paperpi:update_loop:525:DEBUG - checking plugin: Plugin: Word Clock
Feb 23 14:37:47 raspberrypi paperpi[713]: 14:37:47 paperpi:update_loop:528:DEBUG - using pluign: Plugin: Word Clock
Feb 23 14:37:47 raspberrypi paperpi[713]: 14:37:47 paperpi:update_loop:534:DEBUG - screen refresh required
Feb 23 14:37:47 raspberrypi paperpi[713]: 14:37:47 Screen:wrapper:246:DEBUG - initing display
Feb 23 14:37:47 raspberrypi paperpi[713]: 14:37:47 Screen:wrapper:249:DEBUG - Non HD display
Feb 23 14:37:48 raspberrypi paperpi[713]: 14:37:48 epd4in2b_V2:ReadBusy:68:DEBUG - e-Paper busy
Feb 23 14:53:06 raspberrypi systemd[1]: Started Session 5 of user pi.

ejmorgan55, 2022-02-23:
After shutdown and restart on system:
pi@raspberrypi:~/paperpi $ sudo systemctl stop paperpi-daemon; sudo systemctl start paperpi-daemon
pi@raspberrypi:~/paperpi $ tail -f /var/log/daemon.log
Feb 23 15:00:15 raspberrypi paperpi[1564]: 15:00:15 paperpi:main:645:DEBUG - configuration:
Feb 23 15:00:15 raspberrypi paperpi[1564]: {'main': {'display_type': 'epd4in2b_V2', 'vcom': 0.0, 'max_refresh': 4, 'log_level': 'DEBUG', 'splash': 1, 'rotation': 0, 'daemon': True}, 'xPlugin: decimal binary clock': {'layout': 'layout', 'plugin': 'dec_bin_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, 'xPlugin: Weather Adis Ababa': {'layout': 'layout', 'plugin': 'met_no', 'refresh_rate': 300, 'min_display_time': 40, 'max_priority': 2, 'location_name': 'Adis Ababa', 'lat': 9.0, 'lon': 38.75, 'email': 'you@host.diamond'}, 'xPlugin: Crypto Bitcoin v USD': {'plugin': 'crypto', 'layout': 'layout', 'fiat': 'usd', 'coin': 'bitcoin', 'days': 14, 'interval': 'hourly', 'rss_feed': 'https://bitcoinmagazine.com/.rss/full/', 'min_display_time': 30, 'refresh_rate': 300, 'max_priority': 2}, 'xPlugin: Moon Phase': {'layout': 'layout', 'plugin': 'moon_phase', 'min_display_time': 30, 'max_priority': 2, 'email': 'you@host.diamond', 'location_name': 'Europe/Amsterdam', 'lat': 52.3, 'lon': 4.9}, 'xPlugin: A Demo Plugin': {'layout': 'layout', 'plugin': 'demo_plugin', 'min_display_time': 30, 'max_priority': 1, 'your_name': 'Slartybartfast', 'your_color': 'chartreuse'}, 'xPlugin: Basic Clock': {'layout': 'layout', 'plugin': 'basic_clock', 'refresh_rate': 30, 'min_display_time': 50, 'max_priority': 2}, 'xPlugin: XKCD Comic Plugin': {'layout': 'layout', 'plugin': 'xkcd_comic', 'refresh_rate': 1200, 'min_display_time': 120, 'max_x': 800, 'max_y': 600, 'resize': 0, 'max_retries': 10, 'max_priority': 2}, 'xPlugin: Home Assistant': {'layout': 'layout', 'plugin': 'home_assistant', 'min_display_time': 30, 'max_priority': 1, 'home_assistant_basepath': 'https://IP:8123', 'home_assistant_token': '', 'entity1_name': 'Friendly name (ex Indoor)', 'entity1_id': 'home assistant sensor id', 'entity2_name': 'Friendly name (ex Outdoor)', 'entity2_id': 'home assistant sensor id', 'entity3_name': 'Friendly name (ex Outdoor)', 'entity3_id': 'home assistant sensor id', 'entity4_name': 'Friendly name (ex Outdoor)', 'entity5_id': 'home assistant sensor id', 'media_id': 'media player id'}, 'xPlugin: New Yorker Comic': {'layout': 'layout', 'plugin': 'newyorker', 'day_range': 5, 'refresh_rate': 120, 'min_display_time': 60, 'max_priority': 2}, 'xPlugin: Librespot': {'layout': 'layout', 'plugin': 'librespot_client', 'refresh_rate': 10, 'max_priority': 0, 'min_display_time': 15, 'player_name': 'SpoCon-Spotify', 'idle_timeout': 10}, 'xPlugin: Reddit Quotes': {'layout': 'layout', 'plugin': 'reddit_quote', 'refresh_rate': 100, 'min_display_time': 50, 'max_priority': 2, 'max_length': 144}, 'xPlugin: LMS - Your Player Name': {'layout': 'layout', 'plugin': 'lms_client', 'player_name': 'Your Player Name', 'refresh_rate': 5, 'min_display_time': 30, 'max_priority': 0, 'idle_timeout': 15}, 'xPlugin: Word Clock': {'layout': 'layout', 'plugin': 'word_clock', 'refresh_rate': 125, 'min_display_time': 255, 'max_priority': 2}, 'Plugin: Moon Phase': {'layout': 'layout', 'plugin': 'moon_phase', 'min_display_time': 30, 'max_priority': 2, 'email': 'ejmorgan@msn.com', 'location_name': 'America/Phoenix', 'lat': 31.54, 'lon': 110.27}, 'Plugin: Word Clock': {'layout': 'layout', 'plugin': 'word_clock', 'refresh_rate': 125, 'min_display_time': 255, 'max_priority': 2}, '__cmd_line': {'list_plugins': False}}
Feb 23 15:00:15 raspberrypi paperpi[1564]: 15:00:15 Screen:_load_non_hd:442:DEBUG - configuring waveshare_epd.epd4in2b_V2
Feb 23 15:00:15 raspberrypi paperpi[1564]: 15:00:15 Screen:_load_non_hd:472:DEBUG - args_spec: ['self', 'imageblack', 'imagered']
Feb 23 15:00:15 raspberrypi paperpi[1564]: 15:00:15 Screen:epd:332:DEBUG - epd configuration {'epd': <waveshare_epd.epd4in2b_V2.EPD object at 0x761014d0>, 'resolution': [300, 400], 'clear_args': {}, 'one_bit_display': False, 'constants': None, 'mode': '1'}
Feb 23 15:00:16 raspberrypi paperpi[1564]: 15:00:15 epd4in2b_V2:getbuffer:97:DEBUG - Horizontal
Feb 23 15:00:16 raspberrypi paperpi[1564]: 15:00:16 Screen:rotation:379:DEBUG - rotation=0, resolution=[400, 300]
Feb 23 15:00:16 raspberrypi paperpi[1564]: 15:00:16 Screen:wrapper:246:DEBUG - initing display
Feb 23 15:00:16 raspberrypi paperpi[1564]: 15:00:16 Screen:wrapper:249:DEBUG - Non HD display
Feb 23 15:00:16 raspberrypi paperpi[1564]: 15:00:16 epd4in2b_V2:ReadBusy:68:DEBUG - e-Paper busy
Feb 23 15:01:31 raspberrypi rngd[558]: stats: bits received from HRNG source: 80064
Feb 23 15:01:31 raspberrypi rngd[558]: stats: bits sent to kernel pool: 33312
Feb 23 15:01:31 raspberrypi rngd[558]: stats: entropy added to kernel pool: 33312
Feb 23 15:01:31 raspberrypi rngd[558]: stats: FIPS 140-2 successes: 4
Feb 23 15:01:31 raspberrypi rngd[558]: stats: FIPS 140-2 failures: 0
Feb 23 15:01:31 raspberrypi rngd[558]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Feb 23 15:01:31 raspberrypi rngd[558]: stats: FIPS 140-2(2001-10-10) Poker: 0
Feb 23 15:01:31 raspberrypi rngd[558]: stats: FIPS 140-2(2001-10-10) Runs: 0
Feb 23 15:01:31 raspberrypi rngd[558]: stats: FIPS 140-2(2001-10-10) Long run: 0
Feb 23 15:01:31 raspberrypi rngd[558]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Feb 23 15:01:31 raspberrypi rngd[558]: stats: HRNG source speed: (min=999.808; avg=1013.478; max=1021.936)Kibits/s
Feb 23 15:01:31 raspberrypi rngd[558]: stats: FIPS tests speed: (min=25.330; avg=25.714; max=25.950)Mibits/s
Feb 23 15:01:31 raspberrypi rngd[558]: stats: Lowest ready-buffers level: 2
Feb 23 15:01:31 raspberrypi rngd[558]: stats: Entropy starvations: 0
Feb 23 15:01:31 raspberrypi rngd[558]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us

ejmorgan55, 2022-02-23:
sorry; here is the paperpi.log:
pi@raspberrypi:~/paperpi $ sudo nano paperpi.log
  GNU nano 5.4                       paperpi.log *
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:466:D>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:454:I>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 Plugin:_is_ready:230:DEBUG ->
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:461:I>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:464:D>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:465:D>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:466:D>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:454:I>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 Plugin:_is_ready:230:DEBUG ->
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:461:I>
<:464:D>-- Journal begins at Thu 2022-01-27 18:30:20 MST, ends at Wed 2022-02-2>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_loop:504:INFO>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:450:I>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:451:D>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:454:I>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 Plugin:_is_ready:230:DEBUG ->
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:461:I>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:464:D>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:465:D>
Feb 23 14:30:01 raspberrypi paperpi[713]: 14:30:01 paperpi:update_plugins:466:D>

ejmorgan55, 2022-02-23:
Anoterh from 1445:
pi@raspberrypi:~/paperpi $ sudo nano paperpi.log
  GNU nano 5.4                       paperpi.log
-- Journal begins at Thu 2022-01-27 18:30:20 MST, ends at Wed 2022-02-23 15:06:>
Feb 23 14:53:06 raspberrypi sshd[1512]: Accepted password for pi from 192.168.8>
Feb 23 14:53:06 raspberrypi sshd[1512]: pam_unix(sshd:session): session opened >
Feb 23 14:53:06 raspberrypi systemd-logind[488]: New session 5 of user pi.
Feb 23 14:53:06 raspberrypi systemd[1]: Started Session 5 of user pi.
Feb 23 14:58:36 raspberrypi sudo[1546]:       pi : TTY=pts/0 ; PWD=/home/pi/pap>
Feb 23 14:58:36 raspberrypi sudo[1546]: pam_unix(sudo:session): session opened >
Feb 23 14:58:36 raspberrypi systemd[1]: Stopping PaperPi E-Paper Display...
Feb 23 14:59:06 raspberrypi systemd[1]: paperpi-daemon.service: State 'stop-sig>
Feb 23 14:59:06 raspberrypi systemd[1]: paperpi-daemon.service: Killing process>
Feb 23 14:59:06 raspberrypi systemd[1]: paperpi-daemon.service: Killing process>
Feb 23 14:59:06 raspberrypi systemd[1]: paperpi-daemon.service: Main process ex>
Feb 23 14:59:06 raspberrypi systemd[1]: paperpi-daemon.service: Failed with res>
Feb 23 14:59:06 raspberrypi systemd[1]: Stopped PaperPi E-Paper Display.
Feb 23 14:59:06 raspberrypi systemd[1]: paperpi-daemon.service: Consumed 1min 4>
Feb 23 14:59:06 raspberrypi systemd[1]: Started PaperPi E-Paper Display.
Feb 23 14:59:06 raspberrypi sudo[1546]: pam_unix(sudo:session): session closed >
Feb 23 14:59:10 raspberrypi paperpi[1552]: 14:59:10 paperpi:main:643:INFO - ***>
Feb 23 14:59:10 raspberrypi paperpi[1552]: 14:59:10 paperpi:main:645:DEBUG - co>
Feb 23 14:59:10 raspberrypi paperpi[1552]: {'main': {'display_type': 'epd4in2b_

aaronr8684, 2022-02-24:
I just had a (possibly) similar issue running on my Pi4 on 64-bit Bullseye build (PaperPi v0.3). It's not entirely frozen as it still tries to refresh the screen, but nothing is displayed at the end of the cycle. It also doesn't allow me to SSH or VNC into the device, but a local screen works. I'll run through the troubleshooting above before I reboot and see if anything looks off. If it appears to be a different issue, I'll open a new ticket so I'm not hijacking this issue.

aaronr8684, 2022-02-24:
Just kidding. It was a wireless issue.

txoof, 2022-02-24:
@ejmorgan55 
Huh. There's still not much in those logs. Can you try attaching the log as an attachment rather than pasting in? It's a bit easier to look at that way. You can attach files by dragging them and dropping them into the comment box. Git will do the rest for you.

I suspect there's something fishy going on with the SPI bus, but I'm not sure what. Do you have any other things plugged into the GPIO headers? What kind of power supply are you using?

The next time this happens will you try the following:

- Run `sudo raspi-config` and disable SPI and then enable SPI again
- Restart Paperpi `sudo systemctl restart paperpi-daemon`

If the SPI bus is to blame, that should reset it.

txoof, 2022-02-24:
@ejmorgan55 I'm leaning towards this being a hardware issue though I'm keen to figure out what's going on.

Have you successfully used this WaveShare display with any other continuously operating projects?

ejmorgan55, 2022-02-24:
I currently have the device taken apart and I am checking all the connections to ensure it is not a hardware issue; albeit, I have tried 2 different Raspberry Pi and 2 different SD cards. That leaves the wiring connections for any hardware problem. Once I finish checking the wiring; I will reassemble and try again. I do not have anything else connected to this application; simply the Raspberry Pi and the Waveshare 4.2 screen. I am operating it from a 5v 2.5 amp power supply. I thought the power supply might be a problem; however, changing to another power supply; and operating from a USB via my computer reflects the same problems. Also, no underpower indication from the PI. Once reassembled; I will try resetting the SPI. SPI sounds like a good possible cause; albeit, not sure why the SPI would be faulty? I appreciate all the help. It is also interesting to note that the project originally ran for about 24 hours; then froze; then operated about 12 hours for several days; and would come back up after freezing. It then started freezing after about an hour of operation. It then would intermittently reset and come back on. Now, it will not come back up at all; the screen will not even clear. Ergo; looking at the wiring for a possible intermittent connection. FYI, I have this mounted in an old radio. I have an Arduino Nano (operating on a seperate circuit) that plays music for the radio function. The screen added a nice function with the moon phases and word clock, replacing the old tuning panel. As such; I had to simply extend the wiring for the epaper harness between the Raspberry Pi and the display. There could be an intermittent connection; albeit, it seems odd given the nature of problem described above. Regardless; I will post more when I finish troubleshooting. Thanks.

txoof, 2022-02-24:
I've had problems with writing harness in the past. Bad connections resulted in poor image quality.

It sounds like you've eliminated most of the possible problems.

It is also possible that the HAT ha an intermittent fault. It would be great to try another HAT and a screen if you have one.

ejmorgan55, 2022-02-25:
OK; so the wiring harness and hat seem to be fine. I checked all connections. Two things I discovered. One: the paperpi will not start on reboot. You must completely power off and back on. This seems to be a problem with the epaper screen as the program appears to be operating; however, the screen never clears and reinitializes. Once power is removed; it clears and starts fine. Second; the operation seems to be hanging up on the phases of the moon module. After I got the epaper to restart, it froze again several times; each time while displaying the phases of the moon module. I x out the phases of the moon module and operated the system simply using the word clock. It operated great for about 15 hours, no problems. I reassembled the screen and radio; and it started back up with out a problem. There seems to be a problem with phases of the moon updating that is causing the program to freeze. I have a good internet connection; and had the same result on 2 different Pis; as a consequence; I believe it must be either a problem with the program updating or a problem with the remote site. It does not seem to appear on the logs; the display simply freezes and the program continues to operate in the background reporting that the display is busy? I can't find a fault in the program; however, it appears to not update which causes the screen to freeze. Possibly a loop problem where the program tries to continually update the module but cannot? Not sure. What I can say is that it was the phases of the moon module that caused the problem. Once deselected, the program operates just fine. I can live without the phases of the moon and the word clock works well. Not sure if you have any ideas why this might have happened or if you can reproduce it using your set up. Might want to try running phases of the moon and the word clock at the same time to see if you can reproduce the problem on you end. I did note that when I tried the cartoon modules, that frequently, they also would not update and state that a cartoon could not be found; however, the program would continue to function and eventually, a cartoon would be displayed. Again; this sounds like just their server being out of service for a period of time. Thanks for the help; I will operate the system with the word clock as it provide a unique function for the radio restomod.

txoof, 2022-02-26:
@ejmorgan55 There's definitely a bug in the Moon Phases plugin when handling an unreachable end point. Can you search your logs for 'NameError: name 'RequestException' is not defined' and share log lines prior to that error (see the example below).  This is an easy fix. I'll get on it right away.

Can you also post your configuration file? I had a few ideas that I'd like to check out.

```
Feb 26 14:09:00 raspberrypi paperpi[14238]: requests.exceptions.ConnectionError: HTTPSConnectionPool(host='api.met.no', port=443): Max retries exceeded with url: /weatherapi/sunrise/2.0/.json?lat=52.3&lon=4.9&date=2022-02-26&offset=%2B01%3A00 (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x751e3a30>: Failed to establish a new connection: [Errno 111] Connection refused'))
Feb 26 14:09:00 raspberrypi paperpi[14238]: During handling of the above exception, another exception occurred:
Feb 26 14:09:00 raspberrypi paperpi[14238]: Traceback (most recent call last):
Feb 26 14:09:00 raspberrypi paperpi[14238]:   File "paperpi/paperpi.py", line 729, in <module>
Feb 26 14:09:00 raspberrypi paperpi[14238]:   File "paperpi/paperpi.py", line 682, in main
Feb 26 14:09:00 raspberrypi paperpi[14238]:   File "paperpi/paperpi.py", line 431, in build_plugin_list
Feb 26 14:09:00 raspberrypi paperpi[14238]:   File "/tmp/_MEIoXsI1B/library/Plugin.py", line 254, in update
Feb 26 14:09:00 raspberrypi paperpi[14238]:     is_updated, data, priority = self.update_function(*args, **kwargs)
Feb 26 14:09:00 raspberrypi paperpi[14238]:   File "/tmp/_MEIoXsI1B/plugins/moon_phase/moon_phase.py", line 422, in update_function
Feb 26 14:09:00 raspberrypi paperpi[14238]:     except RequestException as e:
Feb 26 14:09:00 raspberrypi paperpi[14238]: NameError: name 'RequestException' is not defined
Feb 26 14:09:00 raspberrypi systemd[1]: paperpi-daemon.service: Main process exited, code=exited, status=1/FAILURE
```

That shouldn't put the display into a frozen state.

ejmorgan55, 2022-02-26:
OK; I applied a fresh download of Raspbian and downloaded the new tar. It is up and running; so far so good. I will let you know if it continues to freeze. Happy that you found the bug in phases of the moon. Not sure why it may have caused the screen to freeze. Thanks.

txoof, 2022-02-27:
@ejmorgan55 I'm not at all convinced that this bug is what is causing your issues. This bug caused PaperPi to crash completely when the API endpoint wasn't available. 

Can you share your configuration file please? You can either attach it or paste it directly into the comment box. 

**Tip** When you paste code/log output, if you surround it with three back tick characters, "\`\`\`" before and after, it will force Gihub to render it as "code" rather than text.

```
Fenced "code block"
with three 
backtick characters
before and 
after
```

unfenced code block
with 
no backtick chars.

ejmorgan55, 2022-02-27:
Not sure which file you want as there are several configure files. Regardless; the paperpi is now running and has not frozen since I downloaded the new Tar. I am not sure why it was freezing; however, it now seems to work fine. I have a 4.2 (B) waveshare screen and changed from the V2 (referenced in the WIKI for the updated B screen) to the BC (also referenced in the WIKI), May have made the difference; as the paperpi has been operating for about 24 hour now without freezing. Here is the configure file:
```#! /bin/sh
# Guess values for system-dependent variables and create Makefiles.
# Generated by GNU Autoconf 2.69 for bcm2835 1.60.
#
# Report bugs to <mikem@airspayce.com>.
#
#
# Copyright (C) 1992-1996, 1998-2012 Free Software Foundation, Inc.
#
#
# This configure script is free software; the Free Software Foundation
# gives unlimited permission to copy, distribute and modify it.
## -------------------- ##
## M4sh Initialization. ##
## -------------------- ##

# Be more Bourne compatible
DUALCASE=1; export DUALCASE # for MKS sh
if test -n "${ZSH_VERSION+set}" && (emulate sh) >/dev/null 2>&1; then :
  emulate sh ```

And the config log

```  GNU nano 5.4                       config.log
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by bcm2835 configure 1.60, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure

## --------- ##
## Platform. ##
## --------- ##

hostname = raspberrypi
uname -m = armv7l
uname -r = 5.10.92-v7+
uname -s = Linux
uname -v = #1514 SMP Mon Jan 17 17:36:39 GMT 2022

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown
  ```