txoof/epd_display #66: executing ./dist/paperpi results in "No such file or directory"

Wiregeek, 2022-03-26:
~/Desktop/paperpi/dist/paperpi exists, is attributed executable, and -rwxrwxrwx 1 pi pi 41389340 Feb 27 21:48 paperpi. 

attempted to execute via terminal from ~, ~/Desktop, ~/Desktop/paperpi, and ~/Desktop/paperpi/dist

Raspberry pi 3b, Waveshare 10.3+ w/ Raspberry hat - confirmed display functional via windows, confirmed interface correct using https://github.com/GregDMeyer/IT8951

Running Raspbian 11, /proc/version says Linux version 5.10.92-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022

Looking for specific advice for locating logs for daemon instance of paperpi, as well as any advice as to what's not doing the thing.

txoof, 2022-03-27:
@Wiregeek  i think I know what the problem is. I'll take a look later today. Thanks for the issue.

txoof, 2022-03-27:
original message:
>running Raspbian (debian 11 bullseye) on a pi 3b+, and I'm not able to start paperpi - when I sudo ./dist/paperpi I get "unable to execute ./dist/paperpi: no such file or directory"

>This is after the install.sh has run successfully, and I'm running it from ~/Desktop/paperpi directory - ./dist/paperpi is a valid location, and paperpi is flagged as executable and I have permissions for it.

>My question to you is where I would go to talk about this, I'm kinda at a loss here. Thank you for any information you can provide or support you can share.

@Wiregeek 
I have an idea of the problem. Can you try the following:

1. Open a terminal and type `ls -alh /usr/bin/paperpi` -- you should see something like this if it's properly installed: `-rwxr-xr-x 1 root root 40M Jan 24 19:42 /usr/bin/paperpi`
2. in the same terminal run `sudo systemctl status paperpi-deamon.service` -- You should see something like this if the daemon is properly installed:
```
$ sudo systemctl status paperpi-daemon.service
● paperpi-daemon.service - PaperPi E-Paper Display
   Loaded: loaded (/etc/systemd/system/paperpi-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2022-03-06 17:21:03 CET; 2 weeks 6 days ago
 Main PID: 9804 (paperpi)
    Tasks: 2 (limit: 3720)
   CGroup: /system.slice/paperpi-daemon.service
           ├─9804 /usr/bin/paperpi -d
           └─9807 /usr/bin/paperpi -d
```
3. Try running `/usr/bin/paperpi` -- that should fire up an instance of paperpi that uses your user config. 

Let us know what you see. If you can paste in any terminal output, that would be helpful.

Wiregeek, 2022-03-27:
pi@raspberrypi:~ $ ls -alh /usr/bin/paperpi
-rwxr-xr-x 1 root root 40M Mar 26 13:58 /usr/bin/paperpi

pi@raspberrypi:~ $ sudo systemctl status paperpi-daemon.service
? paperpi-daemon.service - PaperPi E-Paper Display
     Loaded: loaded (/etc/systemd/system/paperpi-daemon.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sun 2022-03-27 02:30:46 AKDT; 2s ago
    Process: 1732 ExecStart=/usr/bin/paperpi -d (code=exited, status=203/EXEC)
   Main PID: 1732 (code=exited, status=203/EXEC)
        CPU: 7ms

pi@raspberrypi:~ $ /usr/bin/paperpi
bash: /usr/bin/paperpi: No such file or directory
pi@raspberrypi:~ $ sudo /usr/bin/paperpi
sudo: unable to execute /usr/bin/paperpi: No such file or directory

pi@raspberrypi:~ $ /usr/bin/paperconf
a4
pi@raspberrypi:~ $ sudo usr/bin/paperconf
sudo: usr/bin/paperconf: command not found

txoof, 2022-03-27:
Interesting. This indicates that the installer script failed to install the binary into /usr/bin. PaperConf isn't related to this project; you can ignore that.

Can you try to run the installer script again and see what happens. I'll take a look at my end on a fresh install and see if I can't find a similar issue.

txoof, 2022-03-27:
@Wiregeek 
There's definitely something weird about the tar. I'm getting closer to the problem, but I'm not exactly sure what's going on.

Wiregeek, 2022-03-27:
Anything you need tested on hardware let me know, I am monitoring this thread for the next 16 hours

txoof, 2022-03-27:
@Wiregeek I've narrowed down the problem to some difference between how pyinstaller generates executables on different architectures. I don't know exactly what's wrong, but I think I have a solution. I'll ping you when I have something that needs to be tested.

txoof, 2022-03-27:
@aaronr8684 Something very, very strange is going on. I can create a pyinstaller binary with the build scripts that run on the local device. After transferring those binaries to other pis, they refuse to run. I've tried building on both 32bit and 64bit platforms without much luck.

Any ideas?

Wiregeek, 2022-03-27:
That's absolutely bizarre, it doesn't make any sense to me, unless your compile machine has libraries or other required dependencies that the other pis don't, but that's ludicrous.

txoof, 2022-03-27:
@Wiregeek Well, yes. I'm spinning up a fresh buster 32 from scratch and rebuilding the development environment. From there I'll try again. 

Thanks for your patience.

txoof, 2022-03-27:
@Wiregeek If you're feeling zesty, you can try out the development version over here: https://github.com/txoof/PaperPi

It doesn't depend on a pyinstaller frozen blob anymore. We're looking to add gray-scale and color to more devices, but the main branch _should_ be working ok.

aaronr8684, 2022-03-27:
@Wiregeek Can you try running `readelf -l /usr/bin/paperpi` and let me know what the output is? I had a similar issue (#49) when attempting to run the native 32-bit app on a fresh 64-bit build.

Wiregeek, 2022-03-27:
Headed to dev version now. 

**Readelf output**

pi@raspberrypi:~/Desktop/paperpi/dist $ readelf -l /usr/bin/paperpi

Elf file type is EXEC (Executable file)
Entry point 0x114f8
There are 9 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  EXIDX          0x008a18 0x00018a18 0x00018a18 0x00008 0x00008 R   0x4
  PHDR           0x000034 0x00010034 0x00010034 0x00120 0x00120 R   0x4
  INTERP         0x000154 0x00010154 0x00010154 0x00019 0x00019 R   0x1
      [Requesting program interpreter: /lib/ld-linux-armhf.so.3]
  LOAD           0x000000 0x00010000 0x00010000 0x08a24 0x08a24 R E 0x10000
  LOAD           0x008ef8 0x00028ef8 0x00028ef8 0x00268 0x174ac RW  0x10000
  DYNAMIC        0x008f00 0x00028f00 0x00028f00 0x00100 0x00100 RW  0x4
  NOTE           0x000170 0x00010170 0x00010170 0x00044 0x00044 R   0x4
  GNU_STACK      0x000000 0x00000000 0x00000000 0x00000 0x00000 RW  0x10
  GNU_RELRO      0x008ef8 0x00028ef8 0x00028ef8 0x00108 0x00108 R   0x1

 Section to Segment mapping:
  Segment Sections...
   00     .ARM.exidx 
   01     
   02     .interp 
   03     .interp .note.ABI-tag .note.gnu.build-id .gnu.hash .dynsym .dynstr .gnu.version .gnu.version_r .rel.dyn .rel.plt .init .plt .text .fini .rodata .ARM.exidx .eh_frame 
   04     .init_array .fini_array .dynamic .got .data .bss 
   05     .dynamic 
   06     .note.ABI-tag .note.gnu.build-id 
   07     
   08     .init_array .fini_array .dynamic

Wiregeek, 2022-03-27:
@txoof dev branch did the thing and is working, except https://imgur.com/a/vEZataz  which is actually something I've been seeing on everything I've driven this panel with, Windows or Raspbian. I am reading through everything I can find on the pi to see if I can flip horizontal.

txoof, 2022-03-27:
@Wiregeek OHDANG. I've never seen that before. That's really weird. 

Glad the dev version worked for you. That's a relief. Still working on getting pyinstaller blob version working.

aaronr8684, 2022-03-27:
Looks like that's a known issue for the IT8951 displays (see [40](https://github.com/GregDMeyer/IT8951/issues/40) and [43](https://github.com/GregDMeyer/IT8951/pull/43)). 

Does it happen with all images @Wiregeek? We might be able to intercept that issue and just send the image backwards to correct using the built-in PIL ` img.transpose(Image.FLIP_LEFT_RIGHT)`

Wiregeek, 2022-03-27:
yep, happens with everything I've thrown at it - I even moved it to my workstation and got https://imgur.com/a/TCPiSku

txoof, 2022-03-27:
@Wiregeek If you're up for it, would you mind checking out this fixed version: https://github.com/txoof/epd_display/blob/I%2366/paperpi_latest.tgz

Wiregeek, 2022-03-27:
Hokay

After downloading and running install.sh, this result from attempting start of paperpi-daemon.service - this is after doing sudo systemctl restart paperpi-daemon.service and waiting 15 seconds approximate.

pi@raspberrypi:~/Desktop/paperpi $ sudo systemctl status paperpi-daemon.service
? paperpi-daemon.service - PaperPi E-Paper Display
     Loaded: loaded (/etc/systemd/system/paperpi-daemon.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sun 2022-03-27 10:57:30 AKDT; 4s ago
    Process: 2004 ExecStart=/usr/bin/paperpi -d (code=exited, status=203/EXEC)
   Main PID: 2004 (code=exited, status=203/EXEC)
        CPU: 15ms

However, I can at least start and run paperpi itself - I had to manually update the same config file in a few different places, but OK.  Then it runs and starts streaming similar to the below, and is working at that time. 

well, it's still horizontally flipped, but it is working. 

#######
11:00:34 paperpi :_update_plugins:458 :INFO       -      PRIORITY: 2 of max 2
11:00:34 paperpi :_update_plugins:461 :DEBUG      -      DATA: 12 elements
11:00:34 paperpi :_update_plugins:462 :DEBUG      -      IMAGE: <PIL.Image.Image image mode=L size=1872x1404 at 0x7F9E1F8550>
11:00:34 paperpi :update_loop:548 :DEBUG      - plugin data not refreshed, skipping screen update
11:00:39 paperpi :update_loop:495 :INFO       - Plugin: Basic Clock--display for: 38.5 of

txoof, 2022-03-27:
@Wiregeek Great news. Thanks again for testing.

I'll update the repo with the working version.

Wiregeek, 2022-03-27:
I'll go ahead and close, then - I'll open horizontal flip when I get back next weekend, and have something to contribute.

closed 2022-03-27 by Wiregeek

txoof, 2022-03-27:
@Wiregeek Thanks! it would be pretty easy to add that to epdlib and then code in an option to flip the screen horizontal/vertical.

aaronr8684, 2022-03-27:
@Wiregeek could you try putting a file in the bin dir (or where ever you are running PaperPi) called `omni-epd.ini` with the following inside...

```
[Display]
rotate=0  # rotate final image written to display by X degrees [0-360]
flip_horizontal=False  # flip image horizontally
```

I'm not sure if that will work as it stands, but it's worth a shot.