txoof/epd_display #69: Screen Mirrored issue

mTyguy, 2022-12-05:
mirror=True does not reverse mirror for me. I am using 10.3inch EPD and IT8951. here is my config file

# CONFIG_VERSION=1
[main]
# waveshare display type use HD for IT8951 displays
display_type = HD
# required vcom value for IT8951 screens
vcom = -1.50
# maximum refresh between total screen clear for HD displays
max_refresh = 4
# logging output level
log_level = WARNING
# display splash screen (True or False)
splash = True
# rotation of EPD display
# (cable bottom: 0; cable left: -90; cable right: 90; cable top: 180)
rotation = 0
# mirror screen output - useful if output to screen appears backwards
mirror=True

Have also tried:
mirror = True

I'm probably missing something obvious. Any suggestions?

mTyguy, 2022-12-05:
> mirror=True does not reverse mirror for me. I am using 10.3inch EPD and IT8951. here is my config file
>
> `#CONFIG_VERSION=1 [main]
>
> # waveshare display type use HD for IT8951 displays
> display_type = [...]

txoof, 2022-12-05:
@mTyguy

Wow. This feature is utterly broken. Thanks for catching this issue. It should be an easy fix, but will take some diving into the libraries to get there. 

We've archived this form of PaperPi and moved all the development to https://github.com/txoof/PaperPi 

I'll move the issue over there and start working on a fix.